stack-of-tasks/pinocchio #2349: Numpy 2.0 compatibility issue

gfadini, 2024-07-25:
## Bug description

[Some code](https://github.com/stack-of-tasks/pinocchio/blob/c989669e255715e2fa2504b3226664bf20de6fb5/include/pinocchio/bindings/python/context/casadi.hpp#L328) is not compatible with numpy 2.0, see [change guidelines](https://numpy.org/doc/stable//numpy_2_0_migration_guide.html#the-pyarray-descr-struct-has-been-changed).
This leads to a compilation error.
[This change](https://github.com/gfadini/pinocchio/commit/165574e4e3c97659942f986ac635f0203e191249) does the trick, but it is **not** backward compatible.
Luckily though the modification can be made backportable. From the numpy reference:

> New in version NumPy: 2.0 This function was added in a backwards-compatible and backportable way in NumPy 2.0 (see npy_2_compat.h). Any code that previously accessed the ->f slot of the PyArray_Descr, must now use this function and backport it to compile with 1.x. (The npy_2_compat.h header can be vendored for this purpose.)

If you guide me I can add the aforementioned header where seems fit and open a PR.

### System

- OS: Manjaro Linux 24.0.4
- Pinocchio version: master c989669e255715e2fa2504b3226664bf20de6fb5

jcarpent, 2024-07-25:
Thanks @gfadini for raising this issue.
We have already applied some change in EigenPy to be compatible with NumPy 2.x
We will move forward with Pinocchio too.

On your side, do you really need NumPy 2?

gfadini, 2024-07-25:
Hi @jcarpent, I've noticed that eigenpy works without issues, thanks for the change.
I had to make it work with numpy 2.x only due to compatibility with other libraries.
On my side, the library compiles and the tests too.

jcarpent, 2024-08-13:
Solved by #2370