stack-of-tasks/pinocchio #2358: Installation deb pubkey not responding curl command

BarisYazici, 2024-07-31:
## Bug description

[debian 
](http://robotpkg.openrobots.org/packages/debian/robotpkg.asc) server is not responding to the curl command

curl http://robotpkg.openrobots.org/packages/debian/robotpkg.asc 

Adding the user agent string header make it work

curl -A "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36" \
    http://robotpkg.openrobots.org/packages/debian/robotpkg.asc

nim65s, 2024-07-31:
Hi,

This works for me:
```
$ curl http://robotpkg.openrobots.org/packages/debian/robotpkg.asc
-----BEGIN PGP PUBLIC KEY BLOCK-----
...
```

Can you check again to see if this was a transient issue ?

BarisYazici, 2024-07-31:
nope. but could be a network firewall issue. I will try in different network.

closed 2024-07-31 by BarisYazici