stack-of-tasks/pinocchio #2364: MJCF parsing

lvjonok, 2024-08-08:
## Bug description

The `pinocchio` library's `buildModelFromMJCF` function does not correctly parse an MJCF file defining a cart-pole model, resulting in incorrect model parameters. Specifically, the function incorrectly outputs a model with 1 joint and 1 degree of freedom (`model.nq, model.nv` outputs `(1, 1)`) instead of the expected 2 joints and 2 degrees of freedom.

## Expected behavior

The `buildModelFromMJCF` function should correctly parse the MJCF file and recognize both joints (`cart_joint` and `pole_joint`), resulting in a model with 2 joints and 2 degrees of freedom.

## Reproduction steps

Steps to reproduce the behavior:
1. Load the MJCF file:
   ```xml
   <mujoco model="cart_pole">
       <compiler angle="radian" autolimits="true"/>
       <option integrator="RK4"/>
       <statistic meansize="0.262151" extent="0.636603" center="0 0 -0.231699"/>
       <worldbody>
           <body name="cart" pos="0 0 0">
               <inertial pos="0 0 0" mass="1" diaginertia="1e-6 1e-6 1e-6"/>
               <joint name="cart_joint" pos="0 0 0" axis="-1 0 0" type="slide" range="-100 100"/>
               <geom size="0.05 0.05 0.05" type="box" rgba="0 0 0.8 1"/>
               <body name="pole">
                   <inertial pos="0 0 1" mass="1" diaginertia="1e-6 1e-6 1e-6"/>
                   <joint name="pole_joint" pos="0 0 0" axis="0 1 0" range="-600.283 600.283"/>
                   <geom size="0.01 0.25" pos="0 0 0.25" type="cylinder" rgba="0 0 0 1"/>
                   <geom size="0.05" pos="0 0 0.5" rgba="1 0 0 1"/>
               </body>
           </body>
       </worldbody>
       <actuator>
           <motor joint="cart_joint"/>
       </actuator>
   </mujoco>
   ```

2. Parse the MJCF file using pinocchio:
```python
import pinocchio as pin

model = pin.buildModelFromMJCF("assets/cart_pole/model.xml")
data = model.createData()
print(model.nq, model.nv)  # Outputs: (1, 1)
```
3. Observe that the output is (1, 1) instead of the expected (2, 2).
4. Compare the result with MuJoCo:
```python
import mujoco as mj

mjmodel = mj.MjModel.from_xml_path("assets/cart_pole/model.xml")
mjdata = mj.MjData(mjmodel)

print(mjmodel.njnt)  # Should output 2
```

## Additional context

The issue might be due to the unusual format of the MJCF file, which lacks a base link. Adding a base link as shown below allows the file to be parsed correctly:

```xml
<mujoco model="cart_pole">
    <compiler angle="radian" autolimits="true"/>
    <option integrator="RK4"/>
    <statistic meansize="0.262151" extent="0.636603" center="0 0 -0.231699"/>
    <worldbody>
        <body name="base">
            <inertial pos="0 0 0" mass="1" diaginertia="1e-6 1e-6 1e-6"/>
            <body name="cart" pos="0 0 0">
                <inertial pos="0 0 0" mass="1" diaginertia="1e-6 1e-6 1e-6"/>
                <joint name="cart_joint" pos="0 0 0" axis="-1 0 0" type="slide" range="-100 100"/>
                <geom size="0.05 0.05 0.05" type="box" rgba="0 0 0.8 1"/>
                <body name="pole">
                    <inertial pos="0 0 1" mass="1" diaginertia="1e-6 1e-6 1e-6"/>
                    <joint name="pole_joint" pos="0 0 0" axis="0 1 0" range="-600.283 600.283"/>
                    <geom size="0.01 0.25" pos="0 0 0.25" type="cylinder" rgba="0 0 0 1"/>
                    <geom size="0.05" pos="0 0 0.5" rgba="1 0 0 1"/>
                </body>
            </body>
        </body>
    </worldbody>
    <actuator>
        <motor joint="cart_joint"/>
    </actuator>
</mujoco>
```

### System

- Pinocchio version: 3.1.0

jcarpent, 2024-08-08:
Thanks @lvjonok for reporting this issue.
@MegMll Could you have a look to this issue?

jcarpent, 2024-08-26:
Fixed by #2386