stack-of-tasks/pinocchio #2373: Asking about adding the unit description you used in the function, and the order of legs in quadruped example

JuJuManCheng, 2024-08-14:
I would like to ask if the unit you used in quadruped example is meters, quaternion and rediant.
For example, the state 
q1 = np.array( [0.0, 0.0, 0.235, 0.0, 0.0, 0.0, 1.0, 0.8, -1.6, 0.8, -1.6, -0.8, 1.6, -0.8, 1.6])
Also, I would like to check the order of the legs. Is it consistent with the order of joints in URDF file?
Thanks for yout time.

stephane-caron, 2024-08-14:
> I would like to ask if the unit you used in quadruped example is meters, quaternion and rediant.

The floating base joint will be in meters + quaternion, while joint angles are indeed in radians (I have some notes [here](https://stephane-caron.github.io/pink/introduction.html#pink.configuration.Configuration.get_frame_jacobian) in case it helps).

>  Also, I would like to check the order of the legs. Is it consistent with the order of joints in URDF file?

You can check the order of joints to make sure. I believe the quadruped example you are referring to is `anymal-simulation.py`, in which case the model would be an ANYmal B description. You can check joints by printing out the model:

```console
$ python -i load_in_pinocchio.py anymal_b
Robot successfully loaded as <pinocchio.robot_wrapper.RobotWrapper object at 0x7f792c7dca10>
>>> robot.model
Nb joints = 13 (nq=12,nv=12)
  Joint 0 universe: parent=0
  Joint 1 LF_HAA: parent=0
  Joint 2 LF_HFE: parent=1
  Joint 3 LF_KFE: parent=2
  Joint 4 LH_HAA: parent=0
  Joint 5 LH_HFE: parent=4
  Joint 6 LH_KFE: parent=5
  Joint 7 RF_HAA: parent=0
  Joint 8 RF_HFE: parent=7
  Joint 9 RF_KFE: parent=8
  Joint 10 RH_HAA: parent=0
  Joint 11 RH_HFE: parent=10
  Joint 12 RH_KFE: parent=11
```

Here I used [`load_in_pinocchio.py`](https://github.com/robot-descriptions/robot_descriptions.py/blob/df5c76e61ffcf87167c4c9ed371b9661c5a9be70/examples/load_in_pinocchio.py) from [robot_descriptions.py](https://github.com/robot-descriptions/robot_descriptions.py) but you should get the same in the example.

Hoping this helps!

JuJuManCheng, 2024-08-14:
Thanks for your reply! I helps a lot!