stack-of-tasks/pinocchio #2377: unrelated python and c++ examples with the same name

ZeinBarhoum, 2024-08-17:
## Bug description

In the Overview [page](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/) of pinocchio. The `overview-urdf.cpp` and `overview-urdf.py` are completely unrelated. This causes a false impression that doing in c++ what's in python is complicated while it is NOT (especially taking into account the complexity of the existing c++ code for newcomers). 

## Expected behavior

I understand that the overview can include examples of different functionalities in different languages, but the fact that 1) first examples in the page are identical in functionalities between c++ and python, and  2) the existing urdf-examples have the same identical name , make it expected that they both do the same. 
A simple code like this in c++ can be used for example to replicate the python code.

```c++
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include "pinocchio/multibody/fwd.hpp"
#include "pinocchio/parsers/urdf.hpp"

#ifndef PINOCCHIO_MODEL_DIR
#define PINOCCHIO_MODEL_DIR "path_to_the_model_dir"
#endif

int main(int argc, char **argv) {
  // You should change here to set up your own URDF file or just pass it as an
  // argument of this example.
  std::string model_path;
  if (argc < 2) {
    model_path = PINOCCHIO_MODEL_DIR
        "/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf";
  } else {
    model_path = argv[1];
  }

  // Load the urdf model
  pinocchio::Model model;
  pinocchio::urdf::buildModel(model_path, model);
  std::cout << "model name: " << model.name << std::endl;

  // Create data required by the algorithms
  pinocchio::Data data(model);

  // Sample a random configuration
  Eigen::VectorXd q = pinocchio::randomConfiguration(model);
  std::cout << "q: " << q.transpose() << std::endl;

  // Perform the forward kinematics over the kinematic tree
  pinocchio::forwardKinematics(model, data, q);

  // Print out the placement of each joint of the kinematic tree
  for (int i = 0; i < model.names.size(); ++i) {
    std::cout << model.names[i] << " " << data.oMi[i].translation().transpose()
              << std::endl;
  }

  return 0;
}

```

Sorry if this is not the right place for this comment, I started using pinocchio very recently and it's my first time raising an Issue here.

ZeinBarhoum, 2024-08-17:
Apparently they were the same, check the old one [here](https://github.com/stack-of-tasks/pinocchio/blob/0733c368ae0555c155e347fa5ac8e131d9f8e635/examples/overview-urdf.cpp)
The change was made in commit 53272f3

jcarpent, 2024-08-25:
Thanks, @ZeinBarhoum, for raising this issue and pointing out this inconsistency. 
We really appreciate it.
I have just opened this PR (https://github.com/stack-of-tasks/pinocchio/pull/2384) to fix this issue.

jcarpent, 2024-08-25:
Solved via #2384.

closed 2024-08-25 by jcarpent