stack-of-tasks/pinocchio #2380: Runtime issue with pinocchio::urdf::details::parseRootTree function

brolinA, 2024-08-23:
## Bug description

I am using ROS 1 (Noetic) in Ubuntu 20.04 to run the project. The code is able to build using the `catkin build` command. However, when I try to run the code using `rosrun` or `roslaunch` there is a segmentation fault.

## Expected behavior

I followed the [python-ros-example](https://github.com/wxmerkt/pinocchio_ros_example/tree/master) package to build my project. I was able to successfully build and run the [python-ros-example](https://github.com/wxmerkt/pinocchio_ros_example/tree/master) project. But when I tried to use a URDF to load my model it doesn't work.

I also followed the suggestion from issues #1574 and #2321 but neither of them helped. As mentioned before, the **build is successful but fails when I run the code.**  

The build gives the following warning
`/usr/bin/ld: warning: libboost_filesystem.so.1.71.0, needed by /opt/openrobots/lib/libpinocchio_parsers.so.3.0.0, may conflict with libboost_filesystem.so.1.83.0`

However, when I use `dpkg -s libboost-dev | grep 'Version'` to check the version, the output is `Version: 1.71.0.0ubuntu2` which seems to the right version. So, I ignored this warning.

### Code

My minimal ROS Code

```
#include <pinocchio/multibody/fcl.hpp>
#include <pinocchio/parsers/urdf.hpp> 
#include <pinocchio/algorithm/joint-configuration.hpp>
#include <pinocchio/algorithm/kinematics.hpp>
#include <pinocchio/algorithm/geometry.hpp>
#include <pinocchio/algorithm/rnea.hpp>
#include <iostream>
 
int main(int argc, char ** argv)
{

 using namespace pinocchio;
 const std::string urdf_filename = "/home/brolin/pinocchio/models/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf";
 
  // Load the urdf model
  std::cout <<"Urdf path: " << urdf_filename << std::endl;
  Model model;
  pinocchio::urdf::buildModel(urdf_filename, model);
  return 0;
}
```
My `CMakeLists.txt` file
```
cmake_minimum_required(VERSION 3.10)
project(aliengo_dynamics_computer)

## Compile as C++11, supported in ROS Kinetic and newer
add_compile_options(-std=c++14)

## Find catkin macros and libraries
## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
## is used, also find other catkin packages
find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  roscpp
  rospy
  sensor_msgs
  std_msgs
  tf2
  tf2_ros
  message_filters
  message_generation
)

find_package(pinocchio REQUIRED)

catkin_package(
 CATKIN_DEPENDS message_runtime geometry_msgs roscpp rospy sensor_msgs std_msgs tf2 tf2_ros
)

include_directories(
  include
  ${catkin_INCLUDE_DIRS}
  ${PINOCCHIO_INCLUDE_DIRS}
)
link_directories(${PINOCCHIO_LIBRARY_DIRS})

add_executable(urdf_load src/urdf_load.cpp)
target_link_libraries(urdf_load pinocchio::pinocchio)

install(TARGETS urdf_load
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

```

### Screenshots

I used `valgrind` for debugging the code.
[image]

### System

- Ubuntu: 20.04
- Pinocchio: 3.0

nim65s, 2024-08-23:
You have an issue in your apt configuration. Please double check that the things you have in `/etc/apt/sources.list` and `/etc/apt/sources.list.d/*.list` contain only sources for `focal`, which is the code name for 20.04

brolinA, 2024-08-23:
@nim65s Thank you for the response. 

I verified that all the sources in `/etc/apt/souces.list` is for `focal`.

However, inside `/etc/apt/sources.list.d/*` directory, the files named `nvidia-docker.list` and `nvidia-docker.list.save` have something related to `ubuntu 18.04` (shown below). Is this causing the issue? If so, could you please suggest a safe way to remove these files since I have some docker images using these.

```
deb https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/$(ARCH) /
# deb https://nvidia.github.io/libnvidia-container/experimental/ubuntu18.04/$(ARCH) /
deb https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/$(ARCH) /
# deb https://nvidia.github.io/nvidia-container-runtime/experimental/ubuntu18.04/$(ARCH) /
deb https://nvidia.github.io/nvidia-docker/ubuntu18.04/$(ARCH) /
```

nim65s, 2024-08-23:
I think you should remove that and start again with instructions from https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt

nim65s, 2024-08-23:
the important part is removing the broken apt configuration. If this leaves you with broken softwares, then you should decide to either remove those or install them again. To list those, after removing the culprit *.list and `sudo apt update`, you can use `aptitude search '~o'` (you may need a `sudo apt install aptitude` first)

brolinA, 2024-08-23:
I followed the purge command and uninstalled `nvidia-docker` and then did a `apt update`. Then I tried rebuilding the catkin workspace after cleaning it. Now I get the following error.

```
CMake Error at /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: python38) (found version "1.83.0")
Call Stack (most recent call first):
  /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.27/Modules/FindBoost.cmake:2392 (find_package_handle_standard_args)
  /opt/ros/noetic/share/eigenpy/cmake/boost.cmake:144 (find_package)
  /opt/ros/noetic/share/eigenpy/cmake/eigenpyConfig.cmake:149 (SEARCH_FOR_BOOST_PYTHON)
  /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.27/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  /opt/ros/noetic/share/cmake/pinocchio/pinocchioConfig.cmake:160 (find_dependency)
  CMakeLists.txt:22 (find_package)
```

The `CMakeLists.txt:22 (find_package)` refers to the line where I defined `find_package(pinocchio REQUIRED)`

As mentioned before, the boost version that comes when I use the command `dpkg -s libboost-dev | grep 'Version'` is `Version: 1.71.0.0ubuntu2`

nim65s, 2024-08-23:
In your initial post, there are traces of pinocchio installed with robotpkg in /opt/openrobots, and in your current error, there are traces of eigenpy installed with ros in /opt/ros. You can not mix those. I guess in your case, you should `sudo apt-get purge robotpkg-*` too.

This is working for me in a clean dockerfile, so the errors you have are still coming from your setup:
```Dockerfile
FROM ros:noetic

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy --no-install-recommends \
    git \
    ros-noetic-pinocchio

ENV CMAKE_PREFIX_PATH=/opt/ros/noetic
RUN git clone https://github.com/stack-of-tasks/pinocchio-minimal
RUN cmake -B build -S pinocchio-minimal
```

nim65s, 2024-08-23:
I don't know where your boost 1.83 is coming from, but your `dpkg -s libboost-dev | grep 'Version'` is correct, as it is the same as what you should have in `ldd /opt/ros/noetic/lib/libpinocchio.so | grep boost`

brolinA, 2024-08-23:
Just to verify. 

I installed pinocchio by following the instructions from [this page](https://stack-of-tasks.github.io/pinocchio/download.html). Then I also installed it using `apt install ros-noetic-pinocchio`. Can this be causing the issue?

nim65s, 2024-08-23:
The page in your link as several tabs. They are exclusive. You must not mix those. And we should add a ROS tab.

nim65s, 2024-08-23:
If you have the ros-noetic-pinocchio apt package, then you have pinocchio with python.
If you have the robotpkg-py3*-pinocchio apt package, then you have pinocchio with python.

But if you have both, they will break each other.

brolinA, 2024-08-23:
Thank you. That makes sense. I purged `robotpkg` and then tried to build the workspace but I still get the build error mentioned previously. Should I do something after the `purge` command?

> I followed the purge command and uninstalled `nvidia-docker` and then did a `apt update`. Then I tried rebuilding the catkin workspace after cleaning it. Now I get the following error.
>
> ```
> CMake [...]

brolinA, 2024-08-23:
After the `purge` command, I also tried building the `pinocchio_ros_example` package using `catkin build` command and that fails with the same error as well. Any reason why this could be happening?

nim65s, 2024-08-23:
Maybe you need to clear your build cache. I think there is a `catkin clean` command for that, but I'm not sure about the specifics or even which version you have.
If this still does not work, you need to find where this boost is coming from on your system. For that, one solution is `locate boost`. Another is to look into your environment variables and exclude anything outside of `/opt/ros`. You probably also need to check the content of `/usr/local` for software that could cause incompatibilities.

brolinA, 2024-08-23:
@nim65s Thanks. I will search for it.

A quick search using the `locate` command shows that there are some header like the following installed in `miniconda`. Could this be the issue? 

`/home/brolin/miniconda3/pkgs/libboost-headers-1.83.0-ha770c72_0/include/boost/intrusive/slist.hpp`

brolinA, 2024-08-23:
For context, I just tried building the `example.cpp` file from the `pinocchio_ros_example` package using the following command> It was able to build and run successfully. But the package is not being built with the `catkin build` command.

`g++ -std=c++14 example.cpp -o example $(pkg-config --cflags --libs pinocchio)`

nim65s, 2024-08-23:
Yes, once again, you must not use packages installed by different package managers if you are not 100% sure what you are doing.

Using conda is fine, using apt with robotpkg is fine, and using apt with ROS is fine. But they are mutually exclusive, as they can not work together, most of the time. 

You should deactivate your conda environment when you work with ROS packages.

For your `catkin build`, I don't know how you installed that one. On noetic (with a simple `source /opt/ros/noetic/setup.sh` inside a `docker run --rm -it ros:noetic`), I seem to have `catkin_make` and not `catkin`.

jcarpent, 2024-08-24:
I will close an issue as it seems independent of Pinocchio itself.
Thanks to @nim65s for the whole guidance.

closed 2024-08-24 by jcarpent