stack-of-tasks/pinocchio #2412: Issue importing pinocchio.casadi

AFusco99, 2024-09-10:
Hello,
I am installing pinocchio for the first time but I was running into some issues when I tried to execute the examples in the casadi folder you provided. When I try to import adam.casadi I keep receiving this error 
```
    import pinocchio.casadi as cpin
  File "/usr/local/lib/python3.8/site-packages/pinocchio/casadi/__init__.py", line 11, in <module>
    from ..pinocchio_pywrap_casadi import *
ImportError: /usr/local/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap_casadi.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN6casadi6MatrixINS_6SXElemEE9type_nameB5cxx11Ev
```
I have tried to install Pinocchio either building it from robotpk or from source following the instructions. Do you have any suggestions on what could be causing this error?
Thank you

nim65s, 2024-09-10:
Hi @AFusco99,

We already had a report that the casadi feature of pinocchio is not working well with robotpkg. This will probably be fixed by the end of the month, but probably not sooner.

To build pinocchio from source with casadi support, how did you install casadi ?

To avoid building pinocchio from source, but still have casadi support, you should probably try conda or nix.

jcarpent, 2024-09-10:
@AFusco99 If you don't need to interact with other software, I would rather encourage you to install Pinocchio via. 
```bash
conda create -c conda-forge -n pinocchio python=3.12 pinocchio
conda activate pinocchio
python -c "import pinocchio.casadi as cpin"
```

AFusco99, 2024-09-11:
Thank you now it works

closed 2024-09-11 by AFusco99