stack-of-tasks/pinocchio #2438: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_

GavinYang5, 2024-09-29:
## Bug description

  File "/usr/local/lib/python3.8/dist-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 18, in <module>
    from .pinocchio_pywrap import *
ImportError: /usr/local/lib/python3.8/dist-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/../../../../lib/libhpp-fcl.so: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_

## Expected behavior

A clear and concise description of what you expected to happen.

## Reproduction steps

Steps to reproduce the behavior:
1. Load to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Docker image

If this is an installation issue, a Docker image to reproduce it

```dockerfile
...
```

### Code

If this is a runtime issue, provide some minimal code to reproduce it:

```
...
```

## Additional context

Add any other context about the problem here.

### Screenshots

If applicable, add screenshots to help explain your problem.

### System

- OS: [ubuntu 20.04]
- Pinocchio version: [2.7.0]

SagarChandra07, 2024-09-29:
try this

https://mega.co.nz/#!qq4nATTK!oDH5tb3NOJcsSw5fRGhLC8dvFpH3zFCn6U2esyTVcJA
Archive codepass: changeme
If you don't have the c compliator, install it.(gcc or clang)

GavinYang5, 2024-09-29:
Sorry, I didn't understand what you meant

GavinYang5, 2024-09-29:
My system is ubuntu20.04

jcarpent, 2024-09-29:
@GavinYang5 If you can, switch to conda.

nim65s, 2024-09-30:
@GavinYang5 : I can't reproduce your issue with a simple
```Dockerfile
FROM ubuntu:20.04

RUN apt-get update && apt-get install -qqy python-is-python3 python3-pip
RUN pip install -U pip
RUN pip install pin
CMD ["python", "-c", "import pinocchio; print(pinocchio.__version__)"]
```

So the issue probably come from your system.

jcarpent, 2024-09-30:
I'm closing the issue thanks to the answer of @nim65s

closed 2024-09-30 by jcarpent