stack-of-tasks/pinocchio #2444: Boost Version Requirement

Cfather, 2024-10-04:
Hi, I would like to report a tiny issue.

I guess pinocchio >= 3.0.0 is explicitly asking for boost 1.84? since it's calling several new functions in boost::filesystem. The documentation in [https://stack-of-tasks.github.io/pinocchio/download.html](https://stack-of-tasks.github.io/pinocchio/download.html) probably needs to be updated.

I am saying this since I worked a lot in C++. Ubuntu 22.04 installs boost 1.74 by default, so the version problem on boost has been creating some trouble for me.

Since pinocchio installed through conda automatically comes with boost 1.84, one option around could be to manually link the boost libraries inside `/home/{username}/miniconda3/lib`.

nim65s, 2024-10-04:
Hi @Cfather,

What is your issue, exactly ? Do you have any error message ? Can you provide a way to reproduce it ?

Right now, the CI build on 20.04 with boost installed by apt, so I really don't understand your issue.
https://gitlab.laas.fr/stack-of-tasks/pinocchio/-/pipelines/43747

Cfather, 2024-10-04:
Apologize for the lack of information. I guess this is specific to the settings on my computer. I have the following error message:

```shell
/usr/bin/ld: warning: libboost_filesystem.so.1.84.0, needed by /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::stem_v3(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::extension_v3(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::find_parent_path_size(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::append_v3(boost::filesystem::path&, char const*, char const*)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::filename_v3(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/pkgconfig/../../lib/libpinocchio_parsers.so: undefined reference to `boost::filesystem::detail::path_algorithms::find_root_directory(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
```
For example, function `stem_v3` is not introduced in boost 1.74 but introduced in 1.84, while `libpinocchio_parsers.so` is asking explicitly for boost 1.84.

I will close this issue for now and probably come back in the future when I have more information. Thanks for your attention!

closed 2024-10-04 by Cfather