stack-of-tasks/pinocchio #2446: Meshcat-viewer: Scale of .obj meshes gets reset when transformation gets applied

jviereck, 2024-10-04:
I've compiled the latest version of pinocchio from source and the latest meshcat-python package version. The parts of my robot are made of .obj meshes that are scaled to size. When I create a `MeshcatVisualizer` and load the URDF file, the .obj meshes are displayed, however, the scale seems to be ignored.

Spending some time debugging this, it looks like setting the transform

```
            # Update viewer configuration.
            self.viewer[visual_name].set_transform(T)
```

in `MeshcatVisualizer::updatePlacements` is causing the scaling property to be reset. When I apply the scaling once more like this:

```
            # Update viewer configuration.
            self.viewer[visual_name].set_transform(T)

            if visual_name in self.scales:
                self.viewer[visual_name].set_property("scale", self.scales[visual_name])
```

then the scale shows up properly in Meshcat. Here, ` self.scales` is a new dict that stores the scaling values when the objects are created in `MeshcatVisualizer::loadViewerGeometryObject`.

Does this ring a bell for anyone?

PepMS, 2024-11-08:
By adding these lines, it seems that it works:

```py
          Tscale = np.identity(4)
          Tscale[range(3), range(3)] = np.array(scale)
          print(Tscale)

          Ttransform = T @ Tscale

          # Update viewer configuration.
          self.viewer[visual_name].set_transform(Ttransform)
```

jcarpent, 2025-01-12:
Should have been solved in #2511.

closed 2025-01-12 by jcarpent

jviereck, 2025-01-12:
Thank you for fixing this @jcarpent !