stack-of-tasks/pinocchio #2449: Unexpected -O2 flag when building on Debug with Conda

abussy-aldebaran, 2024-10-09:
Hi,

## Bug description

When building pinocchio in Debug, on Ubuntu, using the conda environment installed through 
```
conda env update -n base -f pinocchio/.github/workflows/conda/environment_macos_linux.yml
```
the `-O2` flag is set (among other flags).

## Expected behavior

The `-O2` flag is not set.

## Reproduction steps

It can be seen on pinocchio CI with a failed Debug build job. For example:
https://github.com/stack-of-tasks/pinocchio/actions/runs/11182992289/job/31090536335#step:7:1979

The probable origin of this bug can be seen by executing `env` with the conda environment activated:
```bash
$ env
...
CXXFLAGS=-fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/my_user/miniforge3/envs/pinocchio/include
DEBUG_CXXFLAGS=-fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fvar-tracking-assignments -ffunction-sections -pipe -isystem /home/my_user/miniforge3/envs/pinocchio/include
...
```

## Additional context

While debugging my application using pinocchio, I saw some variables optimized out. After investigating, I found out that my debug build was using the `-O2` flag.
I'm pretty sure it comes from the environment variables set by conda. Searching on the web, I found:
https://github.com/conda/conda/issues/10195
https://github.com/conda/conda-build/issues/3481
https://docs.conda.io/projects/conda-build/en/latest/user-guide/environment-variables.html
But my understanding of conda is too basic to know what to do of it.

So I tried to find if pinocchio's CI was handling this and how, and I found out it isn't.

### Screenshots

From pinocchio CI, https://github.com/stack-of-tasks/pinocchio/actions/runs/11182992289/job/31090536335#step:7:1979
[image]

### System

- OS: Ubuntu with Conda
- Pinocchio version: devel branch

jcarpent, 2024-10-09:
These are the default compilation flags of conda. Have you checked that?
This seems not related to Pinocchio.

abussy-aldebaran, 2024-10-09:
I agree that the problem comes from conda, not Pinocchio. But it seems to me that Pinocchio's CI is impacted.
I'm going to keep investigating to find a clean solution for my project. Could you please let me know if you find one for Pinocchio's CI ?

jcarpent, 2024-10-09:
Yes, you’re correct. I will let @jorisv answer the question.

abussy-aldebaran, 2024-10-09:
I found:
https://github.com/conda-forge/ctng-compiler-activation-feedstock/issues/80
https://github.com/conda-forge/conda-forge.github.io/issues/2002
https://conda-forge.org/docs/user/faq/#faq-compiler-required-options

Apparently, the meta-package `cxx-compiler` is doing too much hand-holding:

> The conda-forge infrastructure provides [activation scripts](https://conda-forge.org/docs/maintainer/adding_pkgs/#activate-scripts) which are run when you conda activate an environment that contains the compiler toolchain. Those scripts set many environment variables that are typically used by GNU autotools and make in the standard (i.e. builtin) build rules. For example, you would see the variable CC set to the long compiler name x86_64-conda-linux-gnu-cc. The activation scripts also set a CMAKE_ARGS variable with many arguments the conda-forge community finds helpful for configuring cmake build flows.

If I understand correctly, a solution would be to install `gxx` and `conda-gcc-specs`. But it wouldn't work for macOs, right ?

jorisv, 2024-10-14:
Hello @abussy-aldebaran,

We are aware of this issue and I find it really annoying.
Like you said, one solution could be to avoid installing cxx-compiler. But this package is really convenient (install the right default compiler per OS).

I'm working to add a pixi support to all our project.
Pixi is a package manager that use conda package but also add a lot of nice feature. On of them is to allow an activation script to run when activating the pixi environment.
In this PR, I use this activation script to unset CXXFLAGS (https://github.com/stack-of-tasks/eigenpy/pull/444/files#diff-963bc921e6166b5aea3f26481499a05fd92db338cdbd106fa49a87845babca71) and avoid user to have this issue.

jorisv, 2024-10-28:
Hello @abussy-aldebaran,

The Pixi PR is merged #2459 

If you use [Pixi](https://pixi.sh/latest/)  to manage your dependencies you will not have this issue anymore.

closed 2024-10-28 by jorisv

abussy-aldebaran, 2024-10-29:
Thanks @jorisv !

I started reading the diff. You removed the conda `.yml` in the github workflows, but I saw a few uses of the variable `$CONDA_PREFIX` in the Pixi scripts. Is that intended ?

jorisv, 2024-11-15:
Yes, I you want to build pinocchio and install it in another environment, then pixi is not tailored to your need. Maybe when `pixi build` will be available, we will have a solution to this use case.

You can still use pixi to generate the environment.yml with all needed dependencies:

```bash
pixi project export conda-environment
```

Then, before building, you can `source development/scripts/pixi/activation.sh` to reset all conda CXX flags.