stack-of-tasks/pinocchio #2451: Error getting to run an example

emrecemal, 2024-10-11:
Hello,

I freshly created a conda environment, and install pinocchio, and I wanted to run this example: examples/cassie-simulation.py
However, I'm getting this error:

```
Traceback (most recent call last):
  File "/home/emrecemal/git_ws/check_cassie/main.py", line 30, in <module>
    robot = pinocchio.RobotWrapper.BuildFromSDF(
  File "/home/emrecemal/git_ws/check_cassie/.venv/lib/python3.10/site-packages/pinocchio/robot_wrapper.py", line 53, in BuildFromSDF
    robot.initFromSDF(
  File "/home/emrecemal/git_ws/check_cassie/.venv/lib/python3.10/site-packages/pinocchio/robot_wrapper.py", line 74, in initFromSDF
    model, constraint_models, collision_model, visual_model = buildModelsFromSdf(
  File "/home/emrecemal/git_ws/check_cassie/.venv/lib/python3.10/site-packages/pinocchio/shortcuts.py", line 113, in buildModelsFromSdf
    model, constraint_models = pin.buildModelFromSdf(
AttributeError: module 'pinocchio.pinocchio_pywrap_default' has no attribute 'buildModelFromSdf'. Did you mean: 'buildModelFromUrdf'?
```

How can I solve it?

MegMll, 2024-10-11:
Hello, 

If you want Sdf support, you need to compile pinocchio from source, because it's not included in the conda package.

jorisv, 2024-11-08:
Hello @emrecemal,

Since Pinocchio 3.3.0, the conda package is build with SDF support.

closed 2024-11-08 by jorisv