stack-of-tasks/pinocchio #2455: CMake must show a build option summary

jorisv, 2024-10-14:
It's sometime difficult to help users when they have issues building Pinocchio.

To win some time, CMake should display the following information:
- OS
- CPU Architecture
- Compiler
- Compiler flags
- Build Type
- Options activated
- Install path
- Tools (doxygen, Python, ...)
  - Binary path
  - Version
- Dependencies
  - Libraries path
  - Include path
  - Version
  - Options

This can help with CI build.

[ViSP](https://github.com/lagadic/visp) is a good example. It provide the following CMake output and also install a file with the following information:

```
│ │ -- ==========================================================
│ │ -- General configuration information for ViSP 3.6.0
│ │ -- 
│ │ --   Version control:               unknown
│ │ -- 
│ │ --   Platform:
│ │ --     Timestamp:                   2024-10-10T14:44:28Z
│ │ --     Host:                        Linux 6.5.0-1025-azure x86_64
│ │ --     CMake:                       3.30.5
│ │ --     CMake generator:             Unix Makefiles
│ │ --     CMake build tool:            $BUILD_PREFIX/bin/make
│ │ --     Configuration:               Release
│ │ -- 
│ │ --   System information:
│ │ --     Number of CPU logical cores: 2
│ │ --     Number of CPU physical cores: 2
│ │ --     Total physical memory (in MiB): 6921
│ │ --     OS name:                     Linux
│ │ --     OS release:                  6.5.0-1025-azure
│ │ --     OS version:                  #26~22.04.1-Ubuntu SMP Thu Jul 11 22:33:04 UTC 2024
│ │ --     OS platform:                 x86_64
│ │ --     CPU name:                    Unknown P6 family
│ │ --     Is the CPU 64-bit?           yes
│ │ --     Does the CPU have FPU?       yes
│ │ --     CPU optimization:            SSE2 SSE3 SSSE3
│ │ -- 
│ │ --   C/C++:
│ │ --     Built as dynamic libs?:      yes
│ │ --     C++ Compiler:                $BUILD_PREFIX/bin/x86_64-conda-linux-gnu-c++  (ver 13.3.0)
│ │ --     C++ flags (Release):         -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-st
│ │ rong -fno-plt -O2 -ffunction-sections -pipe -isystem $PREFIX/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/bld/rattler-build_visp_1
│ │ 728571468/work=/usr/local/src/conda/visp-3.6.0 -fdebug-prefix-map=$PREFIX=/usr/local/src/conda-prefix  -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibil
│ │ ity=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
│ │ --     C++ flags (Debug):           -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-st
│ │ rong -fno-plt -O2 -ffunction-sections -pipe -isystem $PREFIX/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/bld/rattler-build_visp_1
│ │ 728571468/work=/usr/local/src/conda/visp-3.6.0 -fdebug-prefix-map=$PREFIX=/usr/local/src/conda-prefix  -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibil
│ │ ity=hidden -msse2 -msse3 -mssse3 -fPIC -g
│ │ --     C Compiler:                  $BUILD_PREFIX/bin/x86_64-conda-linux-gnu-cc
│ │ --     C flags (Release):           -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -is
│ │ ystem $PREFIX/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/bld/rattler-build_visp_1728571468/work=/usr/local/src/conda/visp-3.6.0 
│ │ -fdebug-prefix-map=$PREFIX=/usr/local/src/conda-prefix  -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DND
│ │ EBUG
│ │ --     C flags (Debug):             -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -is
│ │ ystem $PREFIX/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/bld/rattler-build_visp_1728571468/work=/usr/local/src/conda/visp-3.6.0 
│ │ -fdebug-prefix-map=$PREFIX=/usr/local/src/conda-prefix  -Wall -Wextra -fopenmp -pthread -std=c++17 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
│ │ --     Linker flags (Release):      -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,--allow
[... 135 lines cut ...]
│ │ -- 
│ │ --   DNN: 
│ │ --     Use CUDA Toolkit:            no
│ │ --     Use TensorRT:                no
│ │ -- 
│ │ --   Documentation: 
│ │ --     Use doxygen:                 no
│ │ --     \- Use mathjax:              no
│ │ -- 
│ │ --   Tests and samples:
│ │ --     Use catch2 (built-in):       yes (ver 2.13.7)
│ │ --     Tests:                       yes
│ │ --     Demos:                       yes
│ │ --     Examples:                    yes
│ │ --     Tutorials:                   yes
│ │ --     Dataset found:               no
│ │ -- 
│ │ --   Library dirs:
│ │ --     Eigen3 include dir:          $PREFIX/share/eigen3/cmake
│ │ --     OpenCV dir:                  $PREFIX/lib/cmake/opencv4
│ │ -- 
│ │ --   Install path:                  $PREFIX
│ │ -- 
│ │ -- ==========================================================
```

nim65s, 2024-10-14:
I agree, but I think this is not pinocchio-specific. This could go in jrl-cmakemodules I think.

ManifoldFR, 2024-10-14:
If I may chime in, this is also something that SDL does:
https://github.com/libsdl-org/SDL/blob/3733b1d5d877a8edf0b15a3158c2cb22942a0aab/CMakeLists.txt#L263

They do this by wrapping the CMake `option()` call into a custom function `set_option()` which is defined in their [`macros.cmake`](https://github.com/libsdl-org/SDL/blob/main/cmake/macros.cmake) module.