stack-of-tasks/pinocchio #2458: ImportError: DLL load failed while importing pinocchio_pywrap_default

zifeiyu-c, 2024-10-15:
I use widows10 as the os,the python=3.9.16 and pinocchio=3.1.9.
By the way,i cant get pinocchio by using 'conda install pinocchio -c conda-forge' because it will block when the conda trying 'Solving environment'.
So i directly download the file pinocchio-3.1.0-py39h0bc34c4_2.conda from https://anaconda.org/conda-forge/pinocchio/files and using conda install pinocchio-3.1.0-py39h0bc34c4_2.conda to get pinocchio .
In pinocchio\init.py,when the code run the line of 'from .pinocchio_pywrap_default import *',the error happend.
Author decline in the windows environment i need to add the  pinocchio.dll to the same directory than the .pyd.
But i dont konw how to find the pinocchio.dll and where to add it.
It will be better if authot can give some detailed explanition.

jcarpent, 2024-10-15:
At first glance, it seems your conda env is dirty. 
Please try from a blandly-new env.

I will close this issue as it is not directly related to Pinocchio.

closed 2024-10-15 by jcarpent

zifeiyu-c, 2024-10-15:
What do you mean the conda env is dirty.
Only pinocchio in a new conda environment?

zifeiyu-c, 2024-10-15:
I have create a new environment ,and create a python file only contains 'import pinocchio' by this environment,the error still happens.
Below is the package list for the new environment
ca-certificates           2024.9.24            haa95532_0    defaults
numpy                     2.0.2                    pypi_0    pypi
openssl                   3.0.15               h827c3e9_0    defaults
pinocchio                 3.1.0            py39h0bc34c4_2    <unknown>
pip                       24.2             py39haa95532_0    defaults
python                    3.9.16               h1aa4202_3    defaults
setuptools                75.1.0           py39haa95532_0    defaults
sqlite                    3.45.3               h2bbff1b_0    defaults
tzdata                    2024b                h04d1e81_0    defaults
vc                        14.40                h2eaa2aa_1    defaults
vs2015_runtime            14.40.33807          h98bb1dd_1    defaults
wheel                     0.44.0           py39haa95532_0    defaults

zifeiyu-c, 2024-10-15:
> At first glance, it seems your conda env is dirty. Please try from a blandly-new env.
> 
> I will close this issue as it is not directly related to Pinocchio.

Please check my new comments,tks.

jorisv, 2024-10-15:
Hello,

I have run the following commands on Windows and don't have any issues:

```bash
conda create -n test python=3.9 pinocchio=3.1
conda activate test
python -c "import pinocchio"
```

Something weird in your install is that you have Numpy 2 installed but we didn't release any version of Pinocchio with Numpy 2 support.