stack-of-tasks/pinocchio #2460: MJCF parsing issue for floating base systems

Schulze18, 2024-10-22:
Hi all,

## Bug description

Models of floating base systems are not correctly loaded using `buildModelFromMJCF`.  For example, in quadruped models, the parent link for all the hips is incorrectly set to 0, instead of being assigned to the base or trunk link (which should be 1). As a result, some computations are wrong (crba, nonLinearEffects).

## Expected behavior
The hips' parent link should be the trunk (link 1).

## Reproduction steps

This can be reproduced by loading go1 model from [mujoco-menagerie](https://github.com/google-deepmind/mujoco_menagerie/blob/main/unitree_go1/go1.xml) and compared to loading from [example_robot_data](https://github.com/Gepetto/example-robot-data):
```
import pinocchio as pin
import example_robot_data

# Loading xml file from mujoco menagerie
xml_path = (
 Path(__file__).resolve().parent
  / "go1.xml"
).as_posix()
mjcf_load = pin.buildModelFromMJCF(xml_path)

# Model from example_robot_data
example_data_load = example_robot_data.load("go1").model
```
Inspecting the models:
```
mjcf_load
Nb joints = 14 (nq=19,nv=18)
  Joint 0 universe: parent=0
  Joint 1 root_free: parent=0
  Joint 2 FR_hip_joint: parent=0
  Joint 3 FR_thigh_joint: parent=2
  Joint 4 FR_calf_joint: parent=3
  Joint 5 FL_hip_joint: parent=0
  Joint 6 FL_thigh_joint: parent=5
  Joint 7 FL_calf_joint: parent=6
  Joint 8 RR_hip_joint: parent=0
  Joint 9 RR_thigh_joint: parent=8
  Joint 10 RR_calf_joint: parent=9
  Joint 11 RL_hip_joint: parent=0
  Joint 12 RL_thigh_joint: parent=11
  Joint 13 RL_calf_joint: parent=12
```
```
example_data_load
Nb joints = 14 (nq=19,nv=18)
  Joint 0 universe: parent=0
  Joint 1 root_joint: parent=0
  Joint 2 FL_hip_joint: parent=1
  Joint 3 FL_thigh_joint: parent=2
  Joint 4 FL_calf_joint: parent=3
  Joint 5 FR_hip_joint: parent=1
  Joint 6 FR_thigh_joint: parent=5
  Joint 7 FR_calf_joint: parent=6
  Joint 8 RL_hip_joint: parent=1
  Joint 9 RL_thigh_joint: parent=8
  Joint 10 RL_calf_joint: parent=9
  Joint 11 RR_hip_joint: parent=1
  Joint 12 RR_thigh_joint: parent=11
  Joint 13 RR_calf_joint: parent=12
```

## Additional context

I only tested for quadrupeds (go1 and anymal), but I believe it should also affect humanoids.

Let me know if I'm missing something. Loading manipulators (e.g. Panda) from mujoco menagerie works fine. Thank you in advance.

### System

- Pinocchio version: 3.2.0

MegMll, 2024-10-22:
Hello, 

Thank you for reporting the issue.
It has been fixed (https://github.com/stack-of-tasks/pinocchio/pull/2386) but has not been released yet. It will be in the next release of pinocchio.
In the meantime, if you need to use the mjcf version of go1, i recommend you to  install pinocchio from source, using the version on the devel branch.

Schulze18, 2024-10-24:
Thank you.

Installing from source did the job. I think it was actually solved by [#2403 ](https://github.com/stack-of-tasks/pinocchio/pull/2403), since [#2386](https://github.com/stack-of-tasks/pinocchio/pull/2386) is already in version 3.2.0.

Anyway, thank you again. I'm looking forward to seeing this new release soon.

closed 2024-10-24 by Schulze18