stack-of-tasks/pinocchio #2462: Set -fvisibility=hidden on OSX and GNU/Linux

jorisv, 2024-10-24:
In https://github.com/stack-of-tasks/eigenpy/pull/444 we fall on this specific issue:

When building cppad Python binding on OSX in Debug mode, running the following code will crash:

```python
import pinocchio as pin
import pinocchio.cppad as ADpin
from pycppad import AD, ADFun, Independent
```

It's because pinocchio.cppad doesn't load pycppad as expected.

When building cppad binding, we build the same `module.cpp` than for the double binding.
Then, both binding can have the same symbols defined.
It's what happen in our example: `pinocchio::python::exposeSpecificTypeFeatures` for the double binding is loaded instead of the one from cppad binding. Then pycppad is not imported, the AD scalar type is not registered and the code crash.

To avoid this issue we added `-fvisibility-inlines-hidden` that will set inlined function hidden by default.

But I think we can go further and use [-fvisibility=hidden](https://gcc.gnu.org/wiki/Visibility) on the whole library.
Since we already manage symbol visibility for Windows, I think it will not cost us so much time.

The main advantage:
- Avoid some strange bug
- Smaller binary
  - On https://github.com/stack-of-tasks/eigenpy/pull/444, adding `-fvisibility-inlines-hidden` make the binary go from 1.2Go to 1Go
- Catch bad symbol visibility setup on OSX and GNU/Linux (and avoid debugging that on Windows)

jcarpent, 2024-10-24:
Looks perfect to me. Thanks a lot @jorisv for converging to this suggestion.

jorisv, 2025-01-29:
Closed by #2469

closed 2025-02-14 by jcarpent