stack-of-tasks/pinocchio #2471: Pickling of "pinocchio.pinocchio_pywrap_casadi.Model" instances is not enabled

TexnoMann, 2024-10-31:
## Bug description

Python bindings for the casadi model cannot be serialized via *pickle*.

## Expected behavior

pinocchio.casadi.Model must also be serializable

## Reproduction steps

Install pinocchio==3.2.0 from conda-forge channel and run provided bellow code

### Code
```python

import unittest
import sys
import os
import pickle

import pinocchio as pin
from pinocchio import casadi as cpin

import casadi
from casadi import SX
import numpy as np

class TestMainAlgos(unittest.TestCase):
    def setUp(self):
        self.model = pin.buildSampleModelHumanoidRandom()
        self.data = self.model.createData()

        self.cmodel = cpin.Model(self.model)
        self.cdata = self.cmodel.createData()

    def test_crba(self):
        cmodel = self.cmodel
        cdata = self.cdata
        with open("cmodel.pkl", "wb") as f:
            pickle.dump(cmodel, f)
        with open("cdata.pkl", "wb") as f:
            pickle.dump(cdata, f)

if __name__ == "__main__":
    unittest.main()
```

### System

- OS: Ubuntu 20.04
- Pinocchio version: 3.2.0

jcarpent, 2024-11-01:
Why do yo want to serialize casadi Models? The idea is to have only standard scalar to be serializable

TexnoMann, 2024-11-01:
> Why do yo want to serialize casadi Models? The idea is to have only standard scalar to be serializable
I'm using casadi Models as a backend to calculate filtered dynamics, and my wrapper model that I want to serialize includes cpin.Model. I can write high-level serialization with casadi Model exception if this is expected behavior and you don't plan to implement it.

jcarpent, 2024-11-01:
The shortest way to do it would be to go through a pin.Model and to cast it to a casadi model using 
```python
cmodel = cpin.Model(model)
```

TexnoMann, 2024-11-10:
> The shortest way to do it would be to go through a pin.Model and to cast it to a casadi model using
> 
> ```python
> cmodel = cpin.Model(model)
> ```

Ok, thanks, if there are no plans to serialize symbolic values, the Issue can be closed

jcarpent, 2024-11-11:
It will consume a lot of memory for nothing as we will need to store the computational graph objects behind Casadi types. I would say this should be avoided. @TexnoMann Do you agree?

TexnoMann, 2024-11-17:
> It will consume a lot of memory for nothing as we will need to store the computational graph objects behind Casadi types. I would say this should be avoided. @TexnoMann Do you agree?

Ya, I agree with you. Thank you ;)

closed 2024-11-17 by jcarpent