stack-of-tasks/pinocchio #2480: hpp-fcl to coal migration

v4hn, 2024-11-21:
hpp-fcl just finished it's rebranding to [coal](https://github.com/coal-library/coal/tree/v3.0.0) with version 3.0.

Are there plans when and where to migrate to the new name/version?

jcarpent, 2024-11-30:
@v4hn We will soon update the CMake files of Pinocchio to link directly to coal. I appreciate your understanding.

AndreasKuhner, 2025-03-04:
Hi @jcarpent ,
when do you think you adapt the header files to include the `coal` parts instead of `fcl`? 

Or does `DCOAL_BACKWARD_COMPATIBILITY_WITH_HPP_FCL=ON` already handle warning you receive?

Our problem: We use pinocchio for its collision in the ROS 2 environment and it might be that it treats warnings as errors... 😓 

Cheers,
Andreas

jcarpent, 2025-03-10:
@jorisv Could you answer @AndreasKuhner questions?

jorisv, 2025-03-11:
Hello @AndreasKuhner,

>  when do you think you adapt the header files to include the coal parts instead of fcl?

I can't answer this question. We will switch to coal when some big changed will be merged into pinocchio. I don't know when these change will be ready.

> Our problem: We use pinocchio for its collision in the ROS 2 environment and it might be that it treats warnings as errors... 😓

The hpp-fcl/coal compatibility layer use `#pragma message`. This is not a warning and `-Werror` doesn't convert it into an error (I tested that on g++11).

Can you provide some logs ?

AndreasKuhner, 2025-03-11:
Sure, if you run something which is using Pinocchio via `cmake -DSTRICT=ON  ..`, you will get something like 

```
/opt/openrobots/include/hpp/fcl/coal.hpp:13:2: error: #warning "Please update your includes from 'hpp/fcl' to 'coal'" [-Werror=cpp]
   13 | #warning "Please update your includes from 'hpp/fcl' to 'coal'"
      |  ^~~~~~~
```

aka 'warning' as 'error'

nim65s, 2025-03-11:
My bad, I need to update robotpkg. I'll do thit asap

AndreasKuhner, 2025-03-12:
Thank you guys. This will resolve our issues for now and will be gone eventually after your next bigger release 🥳

jorisv, 2026-04-13:
Done in pinocchio 4

closed 2026-04-13 by jorisv