stack-of-tasks/pinocchio #2486: Installation on Windows wrong.

LiYan-pro, 2024-11-26:
## Bug description
When I want to use `pip install pin` or `conda install pinocchio -c conda-forge` to install pin on Windows with conda, I had never succeed.
Here is my infer, it seems like **pin** needs **Boost** dependency but Boost **can not be install on Windows automatically**, and I don't know how to install this dependency manually.

## Expected behavior
I need a help with how to install **pin** on Windows conda enviroment.

## Additional context
### Screenshots

[image]

### System

- OS: Windows 11
- Pinocchio version: pin==2.7.0

### Error Code

```
Using pip 24.3.1 from E:\miniconda\envs\foundationpose\lib\site-packages\pip (python 3.9)
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Running command pip subprocess to install build dependencies
  Using pip 24.3.1 from E:\miniconda\envs\foundationpose\Lib\site-packages\pip (python 3.9)
  Collecting cmeel-boost~=1.83.0
    Using cached cmeel_boost-1.83.0.tar.gz (4.0 kB)
    Installing build dependencies: started
    Running command pip subprocess to install build dependencies
    Using pip 24.3.1 from E:\miniconda\envs\foundationpose\Lib\site-packages\pip (python 3.9)
    Ignoring numpy: markers 'python_version < "3.9"' don't match your environment
      Link requires a different Python (3.9.20 not in: '>=3.10'): https://files.pythonhosted.org/packages/d1/0f/8d2b5ebb01dc49d20ae0a282d6baff7202b7bf0df8acdd4a6abeffe98070/numpy-2.1.0rc1.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.10)
      Link requires a different Python (3.9.20 not in: '>=3.10'): https://files.pythonhosted.org/packages/54/a4/f8188c4f3e07f7737683588210c073478abcb542048cf4ab6fedad0b458a/numpy-2.1.0.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.10)
      Link requires a different Python (3.9.20 not in: '>=3.10'): https://files.pythonhosted.org/packages/59/5f/9003bb3e632f2b58f5e3a3378902dcc73c5518070736c6740fe52454e8e1/numpy-2.1.1.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.10)
      Link requires a different Python (3.9.20 not in: '>=3.10'): https://files.pythonhosted.org/packages/4b/d1/8a730ea07f4a37d94f9172f4ce1d81064b7a64766b460378be278952de75/numpy-2.1.2.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.10)
      Link requires a different Python (3.9.20 not in: '>=3.10'): https://files.pythonhosted.org/packages/25/ca/1166b75c21abd1da445b97bf1fa2f14f423c6cfb4fc7c4ef31dccf9f6a94/numpy-2.1.3.tar.gz (from https://pypi.org/simple/numpy/) (requires-python:>=3.10)
    Collecting numpy~=1.26.0
      Obtaining dependency information for numpy~=1.26.0 from https://files.pythonhosted.org/packages/b5/42/054082bd8220bbf6f297f982f0a8f5479fcbc55c8b511d928df07b965869/numpy-1.26.4-cp39-cp39-win_amd64.whl.metadata
      Using cached numpy-1.26.4-cp39-cp39-win_amd64.whl.metadata (61 kB)
    Collecting cmeel[build]
      Obtaining dependency information for cmeel[build] from https://files.pythonhosted.org/packages/46/22/774a41b02269f6f0c521cf4fcf5b32536e0906256da6c284a093c7262ce6/cmeel-0.53.3-py3-none-any.whl.metadata
      Using cached cmeel-0.53.3-py3-none-any.whl.metadata (5.1 kB)
    Collecting tomli<3.0.0,>=2.0.1 (from cmeel[build])
      Obtaining dependency information for tomli<3.0.0,>=2.0.1 from https://files.pythonhosted.org/packages/de/f7/4da0ffe1892122c9ea096c57f64c2753ae5dd3ce85488802d11b0992cc6d/tomli-2.1.0-py3-none-any.whl.metadata
      Using cached tomli-2.1.0-py3-none-any.whl.metadata (10.0 kB)
    Collecting cmake<4.0.0,>=3.27.9 (from cmeel[build])
      Obtaining dependency information for cmake<4.0.0,>=3.27.9 from https://files.pythonhosted.org/packages/1f/e3/b5259c734fc91e5ec0cad200e177dbc6434f7f3d414415c119617765b9b1/cmake-3.31.1-py3-none-win_amd64.whl.metadata
      Using cached cmake-3.31.1-py3-none-win_amd64.whl.metadata (6.5 kB)
    Collecting git-archive-all<2.0.0,>=1.23.1 (from cmeel[build])
      Obtaining dependency information for git-archive-all<2.0.0,>=1.23.1 from https://files.pythonhosted.org/packages/62/5f/699a1581fc97e96287f1d1bcbd9757940ff7bc7cc7b1badeff62b8433b48/git_archive_all-1.23.1-py2.py3-none-any.whl.metadata
      Using cached git_archive_all-1.23.1-py2.py3-none-any.whl.metadata (4.3 kB)
    Collecting packaging<24.0,>=23.2 (from cmeel[build])
      Obtaining dependency information for packaging<24.0,>=23.2 from https://files.pythonhosted.org/packages/ec/1a/610693ac4ee14fcdf2d9bf3c493370e4f2ef7ae2e19217d7a237ff42367d/packaging-23.2-py3-none-any.whl.metadata
      Using cached packaging-23.2-py3-none-any.whl.metadata (3.2 kB)
    Collecting wheel>=0.42.0 (from cmeel[build])
      Obtaining dependency information for wheel>=0.42.0 from https://files.pythonhosted.org/packages/0b/2c/87f3254fd8ffd29e4c02732eee68a83a1d3c346ae39bc6822dcbcb697f2b/wheel-0.45.1-py3-none-any.whl.metadata
      Using cached wheel-0.45.1-py3-none-any.whl.metadata (2.3 kB)
    Using cached numpy-1.26.4-cp39-cp39-win_amd64.whl (15.8 MB)
    Using cached cmake-3.31.1-py3-none-win_amd64.whl (36.4 MB)
    Using cached git_archive_all-1.23.1-py2.py3-none-any.whl (10 kB)
    Using cached packaging-23.2-py3-none-any.whl (53 kB)
    Using cached tomli-2.1.0-py3-none-any.whl (13 kB)
    Using cached wheel-0.45.1-py3-none-any.whl (72 kB)
    Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
    Installing collected packages: wheel, tomli, packaging, numpy, git-archive-all, cmake, cmeel
      Creating C:\Users\35964\AppData\Local\Temp\pip-build-env-3zc_6k3r\overlay\Scripts
    Successfully installed cmake-3.31.1 cmeel-0.53.3 git-archive-all-1.23.1 numpy-1.26.4 packaging-23.2 tomli-2.1.0 wheel-0.45.1
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Running command Getting requirements to build wheel
[... 94 lines cut ...]

  Encountered error while generating package metadata.

  See above for output.

  note: This is an issue with the package mentioned above, not pip.
  hint: See above for details.
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: 'E:\miniconda\envs\foundationpose\python.exe' 'E:\miniconda\envs\foundationpose\Lib\site-packages\pip\__pip-runner__.py' install --ignore-installed --no-user --prefix 'C:\Users\35964\AppData\Local\Temp\pip-build-env-1gy8jlge\overlay' --no-warn-script-location --disable-pip-version-check --target '' -v --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cmeel-boost ~= 1.83.0' 'cmeel-urdfdom[build] >= 3.1.1.1, < 4' 'cmeel[build]' 'hpp-fcl[build] >= 2.3.4, < 4'
  cwd: [inherit]
  Installing build dependencies ... error
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

nim65s, 2024-11-26:
pip packaging does not support windows, because I don't know how to work with this system.
I'll let other answer for conda, but I think it is expected to work out of the box.

However, the standard workaround is to use WSL, and use any linux supported install scheme inside windows: conda, nix, pip, robotpkg, or ros.

jcarpent, 2024-11-26:
For conda, have you tried from a clean env?
In this case, could you share the resulting error code? You did only for pip which is not expected to work as mentioned by @nim65s.

jcarpent, 2024-11-30:
I will close this issue as there is no feedback from @LiYan-pro

closed 2024-11-30 by jcarpent