stack-of-tasks/pinocchio #2487: Example meshcat-viewer-octree.py not displaying octree

joao-pm-santos96, 2024-11-26:
Hello developers. 
I've found an issue which I would like to have your comments about. Please, let me know if I can help with anything.

## Bug description

When running examples/meshcat-viewer-octree.py, the script cannot display the octree. Here is the full error:

```bash
This example is skiped as HPP-FCL has not been compiled with octomap support.
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
/home/joao/pixi-test/.pixi/envs/default/lib/python3.12/site-packages/pinocchio/visualize/meshcat_visualizer.py:900: UserWarning: The geometry object named octree is not supported by Pinocchio/MeshCat for vizualization.
  self.loadViewerGeometryObject(
/home/joao/pixi-test/.pixi/envs/default/lib/python3.12/site-packages/pinocchio/visualize/meshcat_visualizer.py:909: UserWarning: The geometry object named octree is not supported by Pinocchio/MeshCat for vizualization.
  self.loadViewerGeometryObject(
gio: http://127.0.0.1:7000/static/: Operation not supported
```

I've also tried installing coal and updating the script to import it instead of hppfcl and the result was:

```bash
Traceback (most recent call last):
  File "/home/joao/pixi-test/../lumen/nexxus/ola.py", line 25, in <module>
    octree_object = pin.GeometryObject("octree", 0, pin.SE3.Identity(), octree)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Boost.Python.ArgumentError: Python argument types in
    GeometryObject.__init__(GeometryObject, str, int, SE3, OcTree)
did not match C++ signature:
    __init__(_object* self, pinocchio::GeometryObject otherGeometryObject)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_texture_path)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_texture_path, boost::variant<pinocchio::GeometryNoMaterial, pinocchio::GeometryPhongMaterial> mesh_material)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_color)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<hpp::fcl::CollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_color, boost::variant<pinocchio::GeometryNoMaterial, pinocchio::GeometryPhongMaterial> mesh_texture_pathmesh_material)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, pinocchio::SE3Tpl<double, 0> placement, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, pinocchio::SE3Tpl<double, 0> placement, std::shared_ptr<hpp::fcl::CollisionGeometry> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path)
[... cut ...] asic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_texture_path, boost::variant<pinocchio::GeometryNoMaterial, pinocchio::GeometryPhongMaterial> mesh_material)
```

### System

- OS: Ubuntu 24.04
- Pinocchio version: 3.3.0
- HPP_FCL version 2.4.5
- Coal version: 3.0.0

nim65s, 2024-11-26:
Hi @joao-pm-santos96,
Coal and hpp-fcl are the same software: please use only one of those, not both.
If you go for coal, for now, you need to add `-DCOAL_BACKWARD_COMPATIBILITY_WITH_HPP_FCL=ON` for pinocchio

joao-pm-santos96, 2024-11-26:
Hi @nim65s 
Yes, I'll do that. I was just trying "random" things out. 
But the fact is that even with hpp-fcl, I was not able to visualize the octree. DO you have any hint on way not?

Thank you!

jorisv, 2024-11-26:
Hello @joao-pm-santos96,

I know we had a lot of regression with all meshact tests. It's really easy to break the meshcat viewer but the unit test are visual, so the CI give us no help.

We will try to fix it.

joao-pm-santos96, 2024-11-26:
Thank you! Let me know if I can help with anything

jorisv, 2024-12-06:
Hello @joao-pm-santos96,

You can check this [PR](https://github.com/stack-of-tasks/pinocchio/pull/2503) to run meshcat-viewer-octree.py.

The `pixi.lock` had been updated, and it depend of coal now. So be sure to clean your pixi environment.

jcarpent, 2024-12-07:
Solved via #2503

closed 2024-12-07 by jcarpent