stack-of-tasks/pinocchio #2504: Python bindings and collision checking not available in ROS 2 Jazzy and Rolling binary installs

sea-bass, 2024-12-07:
## Bug description

I recently engaged in a related issue with collision checking not available in the Humble binaries (https://github.com/stack-of-tasks/pinocchio/issues/2289). The workaround here was to use `target_link_libraries` instead of `ament_target_dependencies` in the `CMakeLists.txt`, which works great for Humble.

However, I just got sent this (https://github.com/sea-bass/pinocchio_ros_cpp_example/issues/1), and can also reproduce it locally.

So it seems like on Ubuntu 24.04, the same problem still persists even when we do `target_link_libraries`?

You can reproduce this with my same example if you go into the `Dockerfile` and replace all mentions of `humble` with `jazzy`.

Interestingly enough, the Jazzy container does not even have the Pinocchio Python bindings installed (I cannot drop into a Python REPL and `import pinocchio`)... so I fear in Ubuntu 24.04, a lot of the add-ons like Python and HPP-FCL may not be actually included?

^ same on ROS 2 Rolling, which uses Pinocchio 3.2.0. No Python bindings or HPP_FCL support, it seems. On this one, I can't even import `pinocchio/collision/collision.hpp` which is supposedly the new header in Pinocchio v3.
## Expected behavior

A clear and concise description of what you expected to happen.

## Reproduction steps

Steps to reproduce the behavior:

1. Go through the steps in https://github.com/sea-bass/pinocchio_ros_cpp_example -- on Humble, this should build fine
2. Replace the `Dockerfile` contents from `humble` to `jazzy` or `rolling` and try again, compiling with fail due to collision checking headers not being available
3. Furthermore, in Jazzy/Rolling, you also will be unable to `import pinocchio` in a Python REPL, indicating that it's also Python bindings that are missing.

### System

- OS: Ubuntu 24.04
- Pinocchio version: 2.6.2 (Jazzy) or 3.2.0 (Rolling)
- Installation: From ROS binaries

sea-bass, 2024-12-07:
@nim65s and @wxmerkt I ended up creating a new issue rather than posting in the old one -- just calling attention to this. Thank you, and happy to work with you to test this out.

nim65s, 2024-12-07:
The humble patch for this was not ported to jazzy:
https://github.com/ros2-gbp/pinocchio-release/blob/patches/release/humble/pinocchio/0003-Patch-Activate-HPP-FCL.patch
https://github.com/ros2-gbp/pinocchio-release/tree/patches/release/jazzy/pinocchio

I'll try to fix this.

sea-bass, 2025-01-04:
As a semi-related update, it appears that `ros-rolling-pinocchio` was removed from the latest Rolling sync (2025-01-02). Likely due to some build failures?

I also noticed that the Jazzy update to these binaries was reverted from `rosdistro` because of OOM issues in the buildfarm?

sea-bass, 2025-04-09:
I think this is now resolved as of the Jazzy sync this week. Thanks, all!

closed 2025-04-09 by sea-bass