stack-of-tasks/pinocchio #2505: when i use pinocchio 3.3.0 on macos, i still get the problem of talos model incorrect loading with the meshcat talos example. what can i do to fix it?

rocketman123456, 2024-12-09:
when i use pinocchio 3.3.0 on macos, i still get the problem of talos model incorrect loading with the meshcat talos example. what can i do to fix it?
my computer is : Macbook Air M3
conda list:
```
# packages in environment at /Users/rocketsky/miniconda3/envs/pinocchio:
#
# Name                    Version                   Build  Channel
absl-py                   2.1.0                    pypi_0    pypi
ampl-asl                  1.0.0                h286801f_2    conda-forge
assimp                    5.4.3                ha9c0b8d_0    conda-forge
asttokens                 3.0.0                    pypi_0    pypi
bzip2                     1.0.8                h80987f9_6    defaults
ca-certificates           2024.11.26           hca03da5_0    defaults
casadi                    3.6.7           py310h3df02a9_0    conda-forge
cli11                     2.4.2                h286801f_0    conda-forge
coal                      3.0.0                hd8ed1ab_5    conda-forge
coal-python               3.0.0           py310hd51063e_5    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
crocoddyl                 2.1.0           py310h7dd35b7_8    conda-forge
decorator                 5.1.1                    pypi_0    pypi
eigen                     3.4.0                h1995070_0    conda-forge
eigenpy                   3.10.1          np20py310h377b2b4_0    conda-forge
etils                     1.11.0                   pypi_0    pypi
example-robot-data        4.1.0              pyh6b898a9_5    conda-forge
exceptiongroup            1.2.2                    pypi_0    pypi
executing                 2.1.0                    pypi_0    pypi
fsspec                    2024.10.0                pypi_0    pypi
gettext                   0.22.5               h8414b35_3    conda-forge
gettext-tools             0.22.5               h8414b35_3    conda-forge
glfw                      2.8.0                    pypi_0    pypi
gmp                       6.3.0                h7bae524_2    conda-forge
hpp-fcl                   3.0.0              pyhd3e5650_4    conda-forge
icu                       75.1                 hfee45f7_0    conda-forge
importlib-resources       6.4.5                    pypi_0    pypi
ipopt                     3.14.16             h3e4dc2c_11    conda-forge
ipython                   8.30.0                   pypi_0    pypi
jedi                      0.19.2                   pypi_0    pypi
libasprintf               0.22.5               h8414b35_3    conda-forge
libasprintf-devel         0.22.5               h8414b35_3    conda-forge
libblas                   3.9.0           25_osxarm64_openblas    conda-forge
libboost                  1.86.0               hc9fb7c5_3    conda-forge
libboost-devel            1.86.0               hf450f58_3    conda-forge
libboost-headers          1.86.0               hce30654_3    conda-forge
libboost-python           1.86.0          py310hf1156d2_3    conda-forge
libboost-python-devel     1.86.0          py310h24597f5_3    conda-forge
libcblas                  3.9.0           25_osxarm64_openblas    conda-forge
libcoal                   3.0.0                hf5bebb4_5    conda-forge
libcxx                    19.1.5               ha82da77_0    conda-forge
libffi                    3.4.4                hca03da5_1    defaults
libgettextpo              0.22.5               h8414b35_3    conda-forge
libgettextpo-devel        0.22.5               h8414b35_3    conda-forge
libgfortran               5.0.0           13_2_0_hd922786_3    conda-forge
libgfortran5              13.2.0               hf226fd6_3    conda-forge
libgz-cmake3              3.5.3                h00cdb27_0    conda-forge
libgz-math7               7.5.1                h5833ebf_2    conda-forge
libgz-tools2              2.0.1                hce4b3f6_0    conda-forge
libgz-utils2              2.2.0                hf9b8971_0    conda-forge
libiconv                  1.17                 h0d3ecfb_2    conda-forge
libintl                   0.22.5               h8414b35_3    conda-forge
libintl-devel             0.22.5               h8414b35_3    conda-forge
liblapack                 3.9.0           25_osxarm64_openblas    conda-forge
liblzma                   5.6.3                h39f12f2_1    conda-forge
libopenblas               0.3.28          openmp_hf332438_1    conda-forge
libosqp                   0.6.3                h5833ebf_1    conda-forge
libqdldl                  0.1.7                hb7217d7_0    conda-forge
libscotch                 7.0.5                hc2b2845_4    conda-forge
libsdformat14             14.5.0               he19963e_0    conda-forge
libsqlite                 3.45.2               h091b4b1_0    conda-forge
libzlib                   1.3.1                h8359307_2    conda-forge
llvm-openmp               19.1.5               hdb05f8b_0    conda-forge
matplotlib-inline         0.1.7                    pypi_0    pypi
meshcat                   0.3.2                    pypi_0    pypi
metis                     5.1.0             h15f6cfe_1007    conda-forge
mujoco                    3.2.6                    pypi_0    pypi
mumps-include             5.7.3                hce30654_5    conda-forge
mumps-seq                 5.7.3                he17653c_5    conda-forge
ncurses                   6.4                  h313beb8_0    defaults
numpy                     2.2.0           py310ha1ddda0_0    conda-forge
octomap                   1.10.0               h7b3277c_0    conda-forge
openssl                   3.4.0                h39f12f2_0    conda-forge
parso                     0.8.4                    pypi_0    pypi
pexpect                   4.9.0                    pypi_0    pypi
pillow                    11.0.0                   pypi_0    pypi
pinocchio                 3.3.0           py310hdd483c8_2    conda-forge
pip                       24.2            py310hca03da5_0    defaults
prompt-toolkit            3.0.48                   pypi_0    pypi
proxsuite                 0.6.7           py310h6854958_2    conda-forge
ptyprocess                0.7.0                    pypi_0    pypi
pure-eval                 0.2.3                    pypi_0    pypi
pygments                  2.18.0                   pypi_0    pypi
pyngrok                   7.2.1                    pypi_0    pypi
pyopengl                  3.1.7                    pypi_0    pypi
python                    3.10.13         h2469fbe_1_cpython    conda-forge
python_abi                3.10                    5_cp310    conda-forge
pyyaml                    6.0.2                    pypi_0    pypi
pyzmq                     26.2.0                   pypi_0    pypi
qhull                     2020.2               h420ef59_5    conda-forge
qhull-static              2020.2               h420ef59_5    conda-forge
readline                  8.2                  h1a28f6b_0    defaults
ruby                      3.2.2                ha6ee62f_1    conda-forge
scipy                     1.14.1          py310hed58976_2    conda-forge
setuptools                75.1.0          py310hca03da5_0    defaults
simde                     0.8.2                h7b3277c_0    conda-forge
sqlite                    3.45.2               hf2abe2d_0    conda-forge
stack-data                0.6.3                    pypi_0    pypi
tinyxml2                  10.0.0               hebf3989_0    conda-forge
tk                        8.6.13               h5083fa2_1    conda-forge
tornado                   6.4.2                    pypi_0    pypi
traitlets                 5.14.3                   pypi_0    pypi
typing-extensions         4.12.2                   pypi_0    pypi
tzdata                    2024b                h04d1e81_0    defaults
u-msgpack-python          2.8.0                    pypi_0    pypi
urdfdom                   4.0.1                h922ef61_0    conda-forge
urdfdom_headers           1.1.2                h7b3277c_0    conda-forge
wcwidth                   0.2.13                   pypi_0    pypi
wheel                     0.44.0          py310hca03da5_0    defaults
xz                        5.4.6                h80987f9_1    defaults
yaml                      0.2.5                h3422bc3_2    conda-forge
zipp                      3.21.0                   pypi_0    pypi
zlib                      1.3.1                h8359307_2    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
```

_Originally posted by @rocketman123456 in https://github.com/stack-of-tasks/pinocchio/issues/1471#issuecomment-2527127727_

jorisv, 2024-12-09:
Hello @rocketman123456,

What example did you run ? `meshcat-viewer.py` ?

rocketman123456, 2024-12-09:
yes，i run a simplify version of meshcat viewer
This is the code I run:

import meshcat
import example_robot_data
import pinocchio
import time
import numpy as np
import casadi
from pinocchio import casadi as cpin
from pinocchio.visualize import MeshcatVisualizer

# load robot model
talos = example_robot_data.load("talos")

viz = MeshcatVisualizer(talos.model, talos.collision_model, talos.visual_model)
viz.initViewer(loadModel=True, open=True)
# Load the robot in the viewer.
viz.loadViewerModel()
viz.displayVisuals(True)
# viz.viewer.jupyter_cell()

# Define a robot configuration (joint positions)
# q = pin.neutral(talos)  # Neutral configuration (all joints at zero)
q0 = talos.q0
v0 = talos.v0

differentiable_talos = cpin.Model(talos.model)
differentiable_data = differentiable_talos.createData()
cpin.forwardKinematics(differentiable_talos, differentiable_data, casadi.SX(q0))

curr_time = 0.0

while True:
    # Generate a simple animation
    for i in range(100):
        q = talos.q0  # pin.neutral(talos)
        q[1] = np.sin(curr_time)  # Example: oscillate a joint
        # curr_time += 0.05
        viz.display(q)
        time.sleep(0.05)  # Pause to simulate real-time
    viz.display(q0)

rocketman123456, 2024-12-09:
It seems that scale of -1.0 does not work when reading urdf file. But I check the code in conda, it should work. I re-create the conda env,but cannot solve the problem

jorisv, 2024-12-10:
Hello @rocketman123456,

Your script is working well on my computer :
[image]

In my case:
- I doesn't install meshcat through pip but with conda `conda install meshcat-python`
- You can also install mujoco from conda (but it's not related to this issue)
- Also unrelated: You don't need to install pinocchio with brew, pinocchio from conda already contains C++ libraries and C++ headers.
   To use it easily in your C++ project you can create a conda environment with `pinocchio`, `cmake`, `pkg-config` and `cxx-compiler`, then cmake will find pinocchio from conda.

Maybe the meshcat from conda and PyPI behave differently, or maybe, your script is loading the pinocchio from brew instead of the pinocchio from conda.
Add this line to your script to check the pinocchio version : `print(pinocchio.__version__)`

rocketman123456, 2024-12-10:
Ah, may be you should take a closer look, the talos right leg and shoulder load incorrectly. It should be symmetrical, but currently the model is overlapped. My code can run on my computer without any warning or error, only this model mesh seems to load with error. I don’t know if this would affect collision calculation or other functions related to this mesh.

jcarpent, 2024-12-10:
@rocketman123456 I will provide a fix soon.

jorisv, 2024-12-10:
Hello @rocketman123456,

Next time try to be more precise in you bug description. I will have been able to help you faster if you tell exactly the issue instead of a really vague « talos model incorrect loading » :/

jcarpent, 2024-12-13:
Fixed in the new release 3.3.1

closed 2024-12-13 by jcarpent

costashatz, 2025-02-25:
@jcarpent @jorisv I am in version pinocchio `3.4.0` and still getting the same issue:

[image]

I installed pinocchio using `pip install pin`. Meshcat version is `0.3.2`. I am using my own Talos URDF, but the model correctly visualizes on all other URDF-based visualizers. Any ideas on what could be wrong?

PR #2511 (2024-12-13): Fix scaling support for meshcat
Fixes #2505