stack-of-tasks/pinocchio #2509: Mesh loading error introduced in Pinocchio 3.3.0

stephane-caron, 2024-12-12:
## Bug description

The following error occurs when running [`humanoid_jvrc.py`](https://github.com/stephane-caron/pink/blob/efd94fb513b51ed354c63cac455fe8f48ed27448/examples/humanoid_jvrc.py) with Pinocchio 3.3.0:

```
/home/tastalian/.micromamba/envs/pink/lib/python3.12/site-packages/pinocchio/visualize/meshcat_visualizer.py:900: UserWarning: Error while loading geometry object: PELVIS_S_0
Error message:
'str' object has no attribute 'is_absolute'
  self.loadViewerGeometryObject(
```

And so forth for each mesh. The result in MeshCat looks like this:

[image]

## Expected behavior

This problem does not occur with Pinocchio 3.2.0, where we get the expected meshes:

[image]

## Reproduction steps

Install pink, robot descriptions and meshcat:

```
conda install -c conda-forge pink robot_descriptions meshcat-python
```

Then run the example script linked above.

## Additional context

### System

- OS: Ubuntu 22.04
- Pinocchio version: 3.3.0

stephane-caron, 2024-12-12:
Upon further investigation https://github.com/stack-of-tasks/pinocchio/pull/2510, this bug is present on conda-forge but has been fixed in the development branch already. I will close this issue when the fix is released.

jcarpent, 2024-12-13:
Fixed in the new release 3.3.1.

closed 2024-12-13 by jcarpent