stack-of-tasks/pinocchio #2519: getFrameAcceleration and getFrameJacobianTimeVariation mismatch?

min-dai, 2024-12-18:
Dear pinocchio developers,

I have a project where I would like to use Pinocchio to do task space control on floating base robots, i.e. 
J*ddq + dJ*dq = 0

According to  #1395, it is suggested to use getFrameAcceleration for dJ*dq term. However, it seems like I got different results on dJ*dq using the acceleration method than getFrameJacobianTimeVariation.

The model is built as:
pinocchio::JointModelComposite jointComposite(2);
jointComposite.addJoint(pinocchio::JointModelTranslation());
iointComposite.addJoint(pinocchio::JointModelSphericalZYX());
pinocchio::urdf::buildModel(urdf_path, jointComposite, model, false);

The update function is:
    pinocchio::forwardKinematics(model, data, q, qdot, Eigen::VectorXd::Zero(model.nv));
    pinocchio::updateFramePlacements(model, data);
    pinocchio::computeJointJacobians(model, data);
    pinocchio::computeJointJacobiansTimeVariation(model, data, q, qdot);

    Eigen::MatrixXd dJacobian = Eigen::MatrixXd::Zero(6, model.nv);
    pinocchio::getFrameJacobianTimeVariation(model, data, frame_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED, dJacobian);
    std::cout << "Time Derivative of Jacobian1: \n" << dJacobian << std::endl;
    std::cout << "dJdq1 : \n" << (dJacobian * qdot).transpose() << std::endl;

    pinocchio::Motion dJdq_motion = pinocchio::getFrameAcceleration(model, data, frame_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED);
    Eigen::VectorXd dJdq(6); 
    dJdq<< dJdq_motion.linear(), dJdq_motion.angular();
    std::cout << "dJdq2: \n" << dJdq.transpose() << std::endl;

In particular, the calculated dJacobian from getFrameJacobianTimeVariation has nonzero entries on base translation, which also seems counterintuitive. 

The return is generated using q and qdot as:
VectorXd q = VectorXd::Zero(robot.get_nq());
q.segment(0, 3) << 0.5, 0.5, 0.5;
VectorXd qdot = VectorXd::Ones(robot.get_nv());

Time Derivative of Jacobian1: 
         0         -1          1    1.78686   -1.11851  -0.668357   -1.11851  -0.328748    2.55569   -1.12018   -1.12018 -0.0647626          0          0          0          0          0          0
         1          0         -1  -0.213136  -0.786866          1  -0.684166    1.20165   0.384243  -0.750314  -0.102666    1.12018          0          0          0          0          0          0
        -1          1          0   0.118504   0.881494  -0.999998   0.778794   -1.11516   0.151602   0.103221  -0.796809   -2.23751          0          0          0          0          0          0
         0          0          0          1         -1          0         -1   0.348019    1.96949   -2.15875   -2.15875          0          0          0          0          0          0          0
         0          0          0         -1          0          1          0   0.810734   -2.15875          0          0    2.15875          0          0          0          0          0          0
         0          0          0          0          1         -1          1   -1.96949   0.348019    1.81073    1.81073         -4          0          0          0          0          0          0
dJdq1 : 
-1.19668  1.16892 -3.11586       -3 0.810734       -1
dJdq2: 
-1.86504 0.382063 -3.23437       -4  1.81073       -2

Are there any problems with the convention or reference frame I used? It would be very helpful if I could get some advice.

jcarpent, 2024-12-19:
I see, you use a Model Joint Composite, which might be the main issue in fact. This is not completed supported yet.

min-dai, 2024-12-19:
In this case, should I resolve the issue by adding floating base joints in urdf?

jcarpent, 2024-12-19:
Or using JointModelFreeFlyer as the last argument of `buildModelFromUrdf(filename, root_joint)`.

closed 2024-12-19 by min-dai

min-dai, 2024-12-19:
Actually, I have a follow-up on these 2 functions where they return unexpected value. I created a test urdf file that only has 2 revolute joints where joint1 is placed at [0.9 0 0] relative to joint0:
```
<robot name="test">

  <link name="world"/>

  <joint name="joint0" type="revolute">
    <parent link="world"/>
    <child link="link1"/>
    <origin xyz="0 0 0" rpy="0 0 0"/>
    <axis xyz="0 0 1"/>
    <limit lower="-10" upper="10" effort="10" velocity="1"/>
  </joint>

  <link name="link1">
    <inertial>
      <mass value="1.0" />
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1" />
      <origin xyz="0.5 0 0" rpy="0 0 0" />
    </inertial>
  </link>

  <joint name="joint1" type="revolute">
    <parent link="link1" />
    <child link="link2" />
    <origin xyz="0.9 0 0" rpy="0 0 0" />
    <axis xyz="0 0 1" />
    <limit lower="-10" upper="10" effort="10" velocity="1" />
  </joint>

  <link name="link2">
    <inertial>
      <mass value="1.0" />
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1" />
      <origin xyz="0.5 0 0" rpy="0 0 0" />
    </inertial>
  </link>

</robot>
```

I then added a frame1 at the same placement as joint1, and performed calculations expecting same results on positions for frame1 and joint1. It turns out frame/joint position, velocities, and Jacobians are consistent and correspond to expected value. However, dJ and dJdq term from acceleration drift is not expected, which can be easily calculated as 
 dJ = [-cos(q0)*L*dq0 0;
           -sin(q0)*L*dq0 0;
         ...] 
and dJdq = [-cos(q0)*L*dq0^2 -sin(q0)*L*dq0^2 ...]

The return was:
[image]

Relavent code:
```
pinocchio::Model model;
    pinocchio::urdf::buildModel(urdf_path, model);
    pinocchio::SE3 placement = pinocchio::SE3::Identity();
    placement.translation() << 0.9, 0, 0;

    pinocchio::JointIndex joint0_id = model.getJointId("joint0");
    pinocchio::JointIndex joint1_id = model.getJointId("joint1");
    model.addFrame(pinocchio::Frame("frame1", joint0_id, 0, placement, pinocchio::OP_FRAME));
    pinocchio::FrameIndex frame1_id = model.getFrameId("frame1");
    //so joint1 and frame1 should have same results
    pinocchio::Data data(model);

    Vector2d q(M_PI/2, 0);
    Vector2d qdot(10,0);

    pinocchio::forwardKinematics(model, data, q, qdot,VectorXd::Zero(model.nv));
    pinocchio::updateFramePlacements(model, data); 
    pinocchio::computeJointJacobians(model, data);
    pinocchio::computeForwardKinematicsDerivatives(model, data, q, qdot, Eigen::VectorXd::Zero(model.nv));
    pinocchio::computeJointJacobiansTimeVariation(model, data, q, qdot);

    double L = 0.9;

        MatrixXd dJacobian_frame1 = Eigen::MatrixXd::Zero(6, model.nv);
    pinocchio::getFrameJacobianTimeVariation(model, data, frame1_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED, dJacobian_frame1);
    std::cout << "Time Derivative of Jacobian frame1: \n" << dJacobian_frame1 << std::endl;
 

    MatrixXd dJacobian_joint1 = Eigen::MatrixXd::Zero(6, model.nv);
    pinocchio::getJointJacobianTimeVariation(model, data, joint1_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED, dJacobian_joint1);
    std::cout << "Time Derivative of Jacobian joint1: \n" << dJacobian_joint1 << std::endl;

    MatrixXd dJacobian_expected = Eigen::MatrixXd::Zero(6, model.nv);
    dJacobian_expected << -cos(q(0))*L*qdot(0), 0, 
    -sin(q(0))*L*qdot(0), 0,
    0,0,
    0,0,
    0,0,
    1,0;
    std::cout << "expected Time Derivative of Jacobian: \n" << dJacobian_expected << std::endl;

    pinocchio::Motion dJdq_frame1 = pinocchio::getFrameAcceleration(model, data, frame1_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED);
    Eigen::VectorXd dJdq(6); // Assuming 6D motion vector
    dJdq<< dJdq_frame1.linear(), dJdq_frame1.angular();
    std::cout << "Frame dJdq from drift:" << dJdq.transpose() << std::endl;

    pinocchio::Motion dJdq_joint1 = pinocchio::getAcceleration(model, data, joint1_id, pinocchio::ReferenceFrame::LOCAL_WORLD_ALIGNED);
    Eigen::VectorXd dJdq_joint(6); // Assuming 6D motion vector
    dJdq_joint<< dJdq_joint1.linear(), dJdq_joint1.angular();
    std::cout << "Joint dJdq from drift:" << dJdq_joint.transpose() << std::endl;

```