stack-of-tasks/pinocchio #2523: Building from source for aarch64

meisbokai, 2024-12-19:
## Bug description
Unable to import pinocchio in Python 3 after a 'successful' build from source. 

## Expected behavior
Import pinocchio into Python 3

## Reproduction steps

Steps to reproduce the behavior:
1. Git clone repository
2. Checkout to devel branch
3. Build (`make -j4`) with:
 - `BUILD_WITH_COLLISION_SUPPORT ON`
 - `BUILD_WITH_COLLISION_SUPPORT ON`
4. `sudo make install`
5. `python`
6. `import pinocchio`

```bash
jetson@ubuntu:~/Documents$ python
Python 3.8.10 (default, Jul 29 2024, 17:02:10) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/pinocchio/__init__.py", line 19, in <module>
    from .pinocchio_pywrap_default import *
SystemError: initialization of pinocchio_pywrap_default raised unreported exception
>>>
```

## Additional context
- Attempting to use Pinocchio in an arm device (Jetson Orin) running on Ubuntu 20.

```bash
jetson@ubuntu:/usr$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
jetson@ubuntu:/usr$ uname -m
aarch64
```

- Not sure if it is relevant, but I am building this together with Crocoddyl 2.1.0 for a project
- Other installation methods:
  - robotpkg - does not have arm64 release
  - pip - Only has version 2.7.0. Require version 3.2.0
  - conda - Not attempted, and not preferred due to other libraries installed via pip for the project. Not recommended to mix package managers (apt, pip and conda)

### System
- Architecture: aarm64
- OS: Ubuntu 20.0.4
- Pinocchio version: 3.2.0

nim65s, 2024-12-19:
Arg, this error message is not helping.
How did you install pinocchio's dependencies ?
Can you show the result of `ls -l /usr/local/lib/python3.8/site-packages/pinocchio/` and `ldd /usr/local/lib/python3.8/site-packages/pinocchio/*.so` ?

NB: 
- pip package version should be updated by the end of the year. 
- nix package is available and imports correctly on arm64-linux. And crocoddyl is available directly too. But, as with conda, this will provide another python interpreter (3.11 or 3.12), and not be compatible with the python from apt (3.8)
- ros package is expected to be searchable when attempting to build crocoddyl. You probably only need something like `export CMAKE_PREFIX_PATH=/opt/ros/your-ros-version`

jorisv, 2024-12-19:
Hello @meisbokai,

Can you display the first cmake output (remove your build directory, relaunch cmake). So we can see which compiler you are using.

conda can manage pip dependency, you first have to install pip : `conda install pip`.

meisbokai, 2024-12-20:
@nim65s Thank you for your response, and my apologies for the missing details. 

- Pinocchio's dependencies were either built from source, or installed via apt
  - eigenpy -> Built from source v3.9.0
  - requires HPP-FCL -> Built from source v2.4.5
    - requires assimp -> `sudo apt-get install libassimp-dev`
  - requires urdfdom_header -> `sudo apt install liburdfdom-headers-dev`
  - requires urdfdom -> `sudo apt install liburdfdom-dev`
  - requires Qhull -> Built from source

- I will be looking forward to the updated pip package version!
- I would not be able to use Crocoddyl directly as my project involves modifying Crocoddyl's source code.

- Below are the results of the commands.
```bash
jetson@ubuntu:~$ ls -l /usr/local/lib/python3.8/site-packages/pinocchio/
total 50168
-rw-r--r-- 1 root staff      207 Sep 17 12:49 deprecated.py
-rw-r--r-- 1 root staff     1085 Sep 17 12:49 deprecation.py
drwxr-sr-x 2 root staff     4096 Dec 20 08:44 derivative
-rw-r--r-- 1 root staff     1192 Sep 17 12:49 explog.py
-rw-r--r-- 1 root staff     2270 Sep 17 12:49 __init__.py
lrwxrwxrwx 1 root staff       62 Dec 20 08:44 pinocchio_pywrap_default.cpython-38-aarch64-linux-gnu.so -> pinocchio_pywrap_default.cpython-38-aarch64-linux-gnu.so.3.2.0
-rw-r--r-- 1 root staff 51300496 Dec 20 02:21 pinocchio_pywrap_default.cpython-38-aarch64-linux-gnu.so.3.2.0
-rw-r--r-- 1 root staff    14771 Oct 11 09:38 robot_wrapper.py
-rw-r--r-- 1 root staff     2494 Sep 17 12:49 romeo_wrapper.py
-rw-r--r-- 1 root staff     8908 Sep 17 12:49 shortcuts.py
-rw-r--r-- 1 root staff     2445 Sep 17 12:49 utils.py
drwxr-sr-x 2 root staff     4096 Dec 20 08:44 visualize
-rw-r--r-- 1 root staff     1556 Sep 17 12:49 windows_dll_manager.py
jetson@ubuntu:~$ ldd /usr/local/lib/python3.8/site-packages/pinocchio/*.so
	linux-vdso.so.1 (0x0000ffff817bd000)
	libeigenpy.so => /usr/local/lib/libeigenpy.so (0x0000ffff7084d000)
	libboost_python38.so.1.71.0 => /lib/aarch64-linux-gnu/libboost_python38.so.1.71.0 (0x0000ffff707b6000)
	libpinocchio_parsers.so.3.2.0 => /usr/local/lib/libpinocchio_parsers.so.3.2.0 (0x0000ffff7072f000)
	libconsole_bridge.so.0.4 => /lib/aarch64-linux-gnu/libconsole_bridge.so.0.4 (0x0000ffff7071a000)
	liburdfdom_world.so.1.0 => /lib/aarch64-linux-gnu/liburdfdom_world.so.1.0 (0x0000ffff706e9000)
	libpinocchio_collision.so.3.2.0 => /usr/local/lib/libpinocchio_collision.so.3.2.0 (0x0000ffff706d3000)
	libhpp-fcl.so => /usr/local/lib/libhpp-fcl.so (0x0000ffff6ff2f000)
	libpinocchio_extra.so.3.2.0 => /usr/local/lib/libpinocchio_extra.so.3.2.0 (0x0000ffff6ff04000)
	libpinocchio_default.so.3.2.0 => /usr/local/lib/libpinocchio_default.so.3.2.0 (0x0000ffff6fa8b000)
	libboost_serialization.so.1.71.0 => /lib/aarch64-linux-gnu/libboost_serialization.so.1.71.0 (0x0000ffff6fa3a000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff6f855000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff6f7aa000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff6f786000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff6f613000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff8178d000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff6f5e2000)
	libboost_filesystem.so.1.71.0 => /lib/aarch64-linux-gnu/libboost_filesystem.so.1.71.0 (0x0000ffff6f5b8000)
	liburdfdom_model.so.1.0 => /lib/aarch64-linux-gnu/liburdfdom_model.so.1.0 (0x0000ffff6f587000)
	libtinyxml.so.2.6.2 => /lib/aarch64-linux-gnu/libtinyxml.so.2.6.2 (0x0000ffff6f561000)
	libassimp.so.5 => /lib/aarch64-linux-gnu/libassimp.so.5 (0x0000ffff6e9d6000)
	liboctomap.so.1.9 => /lib/aarch64-linux-gnu/liboctomap.so.1.9 (0x0000ffff6e97f000)
	liboctomath.so.1.9 => /lib/aarch64-linux-gnu/liboctomath.so.1.9 (0x0000ffff6e969000)
	libqhull_r.so.8.1 => /usr/local/lib/libqhull_r.so.8.1 (0x0000ffff6e8ea000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff6e8c0000)
	libminizip.so.1 => /lib/aarch64-linux-gnu/libminizip.so.1 (0x0000ffff6e8a6000)
```

@jorisv Thank you for your response, as well as your tip on conda. Below is the cmake output.

```
-- JRL cmakemodules found in 'cmake/' git submodule
-- Configuring "pinocchio" (http://github.com/stack-of-tasks/pinocchio)
-- Package version (ROS package.xml): 3.2.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
[... 55 lines cut ...]
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.71.0") found components: chrono thread date_time serialization filesystem atomic 
-- C++ standard sufficient: Minimal required 11, currently defined: 14
-- Performing Test res_-Wno-conversion
-- Performing Test res_-Wno-conversion - Success
-- Performing Test res_-Wno-comment
-- Performing Test res_-Wno-comment - Success
-- Performing Test res_-Wno-self-assign-overloaded
-- Performing Test res_-Wno-self-assign-overloaded - Failed
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework 
-- Valgrind not found, memory checks are disabled
-- Performing Test res_-Wno-maybe-uninitialized
-- Performing Test res_-Wno-maybe-uninitialized - Success
-- Performing Test res_-Wuse-after-free=0
-- Performing Test res_-Wuse-after-free=0 - Failed
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetson/Documents/Docker_Bokai/workspaces/new_pepms/pinocchio/build
```

nim65s, 2024-12-20:
Ok, I can not see anything wrong in your setup.
We might need to start a debugger to dig further.

> * I would not be able to use Crocoddyl directly as my project involves modifying Crocoddyl's source code.

crocoddyl, since https://github.com/loco-3d/crocoddyl/pull/1296, is a flake ; so you can modify it as you wish and use `nix build` or `nix run` to download automatically all its dependencies from the binary cache and build the package with your modified sources.

jcarpent, 2024-12-20:
@meisbokai have you tried using `conda install pinocchio`? We do support aarch64 normally