stack-of-tasks/pinocchio #2524: Undefined: WIN32, __builtin_assume, __builtin_addressof

OoLoueLoueoO, 2024-12-20:
## Bug description

I'm trying to use Pinocchio in Simulink on Windows with Simulink's [custom code functionality](https://www.mathworks.com/help/stateflow/ug/c-custom-code-library-charts.html). 
The compiler is configured to be MinGW64 and header and lib path are configured the same as previously on Linux, which I have compilation working and the executable was tested and running. 
The Pinocchio is installed using conda. 

## Expected behavior

When compile the code, the following error shows up. (undefined WIN32, __builtin_assume, __builtin_addressof)

> Error while parsing custom code for model '[pino_simulink_test](matlab:open_system ('pino_simulink_test'))'. Fix the errors in your custom code or disable '[Import custom code](matlab:SLCC.Utils.OpenConfigureSetAndHighlightParseCC('pino_simulink_test'))' if your custom code has incompatibilities. If your custom code uses C++ language syntax, ensure that 'Language' is set to 'C++' in Model Configuration Parameters > Simulation Target. 
> Caused by:
> In line 150 of [c:\users\haowenl\appdata\local\anaconda3\library\include\pinocchio\macros.hpp](matlab: opentoline('c:%5cusers%5chaowenl%5cappdata%5clocal%5canaconda3%5clibrary%5cinclude%5cpinocchio%5cmacros.hpp', 150)),
> expected an expression:
> |  #if WIN32
> |           ^
> In line 67 of [c:\users\haowenl\appdata\local\anaconda3\library\include\boost\smart_ptr\detail\local_counted_base.hpp](matlab: opentoline('c:%5cusers%5chaowenl%5cappdata%5clocal%5canaconda3%5clibrary%5cinclude%5cboost%5csmart_ptr%5cdetail%5clocal_counted_base.hpp', 67)),
> identifier "__builtin_assume" is undefined:
> |          __builtin_assume( local_use_count_ >= 1 );
> |          ^
> In line 40 of [c:\users\haowenl\appdata\local\anaconda3\library\include\boost\core\addressof.hpp](matlab: opentoline('c:%5cusers%5chaowenl%5cappdata%5clocal%5canaconda3%5clibrary%5cinclude%5cboost%5ccore%5caddressof.hpp', 40)),
> identifier "__builtin_addressof" is undefined:
> |      return __builtin_addressof(o);
> |             ^
> |          detected during:
> |            instantiation of "T *boost::addressof(T &) [with T=const pinocchio::GeometryModel::CollisionPairVector]" at line 649 of "c:\users\haowenl\appdata\local\anaconda3\library\include\boost\foreach.hpp"
> |            instantiation of "boost::foreach_detail_::auto_any<T *> boost::foreach_detail_::contain(T &, mpl_::false_ *) [with T=const pinocchio::GeometryModel::CollisionPairVector]" at line 268 of
> 

## Reproduction steps

Steps to reproduce the behavior:
1. compile the code in Simulink
2. or use "validate custom code" button when setting up the Simulink environment. 

### Simulink Environment parameters:
Include Headers:
#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/model.hpp"
#include "load_robot_c_caller.hpp"

Include Directories:
C:\Users\haowenl\AppData\Local\anaconda3\Library\include
C:\Users\haowenl\AppData\Local\anaconda3\Library\include\eigen3

Source Files:
load_robot_c_caller.cpp

Libraries:
C:\Users\haowenl\AppData\Local\anaconda3\Library\lib\pinocchio_default.lib
C:\Users\haowenl\AppData\Local\anaconda3\Library\lib\pinocchio_parsers.lib

Compiler flags:
-DNOMINMAX

Linker Flags:
-lpinocchio_default -lpinocchio_parsers 

### System

- OS: [windows 11]
- Pinocchio version:  [3.1.0]

jorisv, 2024-12-20:
Hello @OoLoueLoueoO,

I don't know if Pinocchio can be build with MinGW, we don't have a CI setup to test that configuration.
Conda-forge Pinocchio package is build with MSVC compilers and I'm not sure MinGW and MSVC binaries are compatible. Is there a way to tell simulink to use MSVC compiler ?

Also, I see this message in the log you provided:

> If your custom code uses C++ language syntax, ensure that 'Language' is set to 'C++' in Model Configuration Parameters > Simulation Target.

Did you choose C++ language ? Pinocchio is a C++ library.

OoLoueLoueoO, 2024-12-31:
Thank for your response jorisv. It does seems like a compiler config issue.  I was able to compile and generate the model and data using gcc in the MATLAB (in ubuntu) right now.

closed 2024-12-31 by OoLoueLoueoO