stack-of-tasks/pinocchio #2525: Simplify pinocchio.pc (pkg-config)

jorisv, 2024-12-20:
Pinocchio pkg-config file have a lot of drawback:

- It doesn't allow to choose a particular pinocchio target
- Code to generate it is redundant with the cmake target definition
- Some dependencies like hpp-fcl doesn't provide .pc file anymore
- jrl-cmakemodules code to manage .pc file generation is not maintained

I plan to simplify it by only allowing to link against pinocchio_default.

I don't know what pkg-config use case is. So if you use it, please, comment on this issue to expose your use case.

nim65s, 2024-12-20:
Also, boost does not provide a .pc, so our .pc are doomed to be incomplete, unless we generate the required boost.pc ourself.

I would advocate for a simple removal of pkg-config support from all our codebases, if nobody provide a convincing use case 👼🏼

ManifoldFR, 2024-12-20:
I know there's been discussions about getting rid of pkg-config support - I also support it

jcarpent, 2024-12-20:
On my side, I've just use pkg-config for quick compilation via g++:
```bash
g++ -O3 -std=c++14 -g overview-urdf.cpp -o overview-urdf $(pkg-config --cflags pinocchio) $(pkg-config --libs pinocchio) && ./overview-urdf
```