stack-of-tasks/pinocchio #2527: closed-loop mechanisms

shuangxxxx, 2024-12-23:
Hello, I am actually working on closed-loop mechanisms. I noticed that the example provided in the Pinocchio package includes a simulation for closed-kinematic-chains (simulation-closed-kinematic-chains.py), but when I run it, I encounter the following issue:

```
placement_center_link_A.translation = pin.XAxis * length_link_A / 2.0
AttributeError: module 'pinocchio' has no attribute 'XAxis'
```
Additionally, I saw that Pinocchio 3.0 has updates for supporting closed-loop mechanisms. Could you point me to any documentation or resources related to this?
### System

- OS:ubuntu 22.04
- Pinocchio version: 3.3.1

jcarpent, 2024-12-23:
I'm not able to reproduce your issue.
Could you tell us how did you install Pinocchio?
Could you show the result of in your terminal:
```bash
python -c "import pinocchio as pin; print('version:', pin.__version__); print('install:',pin.__file__)"
```

shuangxxxx, 2024-12-23:
I have installed Pinocchio on Conda.
[image]

jcarpent, 2024-12-23:
It seems you have local troubles (core dumped). Please install Pinocchio in a fresh conda env and check whether you have multiple installation of Pinocchio before running again the bash line above.

shuangxxxx, 2024-12-24:
Thank you for your reply. After uninstalling Pinocchio from ROS, it works normally.

In the `simulation-closed-kinematic-chains.py` example, the motion of a four-bar linkage under gravity is simulated. All joint positions are initialized, and then updated based on constraints.

Is it possible to compute all joint positions/velocities using only the position/velocity of one joint (e.g., `link_B1`) while satisfying the constraints?