stack-of-tasks/pinocchio #2531: Wrong velocity of the center of mass

hang0610, 2024-12-24:
## Bug description

Hi, there. Thanks for making this great repo public. Recently I try to use Pinocchio to compute the vel and acc of center of mass for a floating base robot. But the direction of velocity computed by `pin.centerOfMass` seems to be opposite to the actual velocity. Here is what I did:

### Import the Model
```python
self.robot = pin.RobotWrapper.BuildFromURDF(
            self.config["ASSET_ROOT"] + '/' + self.config["ASSET_FILE"],
            self.config["ASSET_ROOT"]
        )
self.mj_model = mujoco.MjModel.from_xml_path(self.config["ROBOT_SCENE"])
self.mj_data = mujoco.MjData(self.mj_model)
```
### Compute CoM
```python
def calculate_com(self, visualize=False):
        self.robot.model.gravity.linear = np.array([0, 0, -9.81])
        pin.centerOfMass(self.robot.model, self.robot.data, self.mj_data.qpos, self.mj_data.qvel)
        if visualize:
            # Visualize COM
            self.mj_model.site_pos[self.mj_model.site("com_marker").id] = self.robot.data.com[0]
        print("qvel[0:3]: ", self.mj_data.qvel[:3], " v_CoM: ", self.robot.data.vcom[0])
        return self.robot.data.com[0], self.robot.data.vcom[0], self.robot.data.acom[0]
```
## Expected behavior
Here, the robot will be lifted up and the `qvel[2]` and `self.robot.data.vcom[0][2]` shoule both be positive, just like this

https://github.com/user-attachments/assets/fc60b56a-ad12-41fe-9b57-29f562ddda61

The red ball in the middle of the torso is the visualized CoM. However, It turns out that the `qvel[2]` is positive while `self.robot.data.vcom[0][2]` is **Negative**.

### Screenshots

[image]

### System

- OS: IOS
- Pinocchio version: 2.6.20

Thanks in advance!

jcarpent, 2024-12-27:
@hang0610 Could you share a full script to reproduce your issue locally, including the robot model file?

hang0610, 2024-12-27:
Yes, sure. The `.zip` file below contains the minimum script and models to reproduce my issue.

[test.zip](https://github.com/user-attachments/files/18264260/test.zip)

To reproduce, just run:
```bash
# Linux
python test.py

# Mac
mjpython test.py
```

Thanks for your time!

hang0610, 2024-12-30:
I found the problem: mujoco is `wxyz` but pinocchio is `xyzw`. I notice this when I visualize my robot in meshcat with `wxyz` and it looks like this😂:
[image]

closed 2024-12-30 by hang0610