stack-of-tasks/pinocchio #2536: STD exception 'boost::wrapexcept<boost::bad_get>': 'boost::bad_get: failed value get using boost::get' was caught.

OoLoueLoueoO, 2024-12-31:
## Bug description

A clear and concise description of what the bug is.

## Expected behavior

A clear and concise description of what you expected to happen.

## Reproduction steps

Steps to reproduce the behavior:
1. Load to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Docker image

If this is an installation issue, a Docker image to reproduce it

```dockerfile
...
```

### Code

If this is a runtime issue, provide some minimal code to reproduce it:

```
...
```

## Additional context

Add any other context about the problem here.

### Screenshots

If applicable, add screenshots to help explain your problem.

### System

- OS: [e.g. iOS]
- Pinocchio version: [e.g. 0.1.0]

jorisv, 2025-01-02:
Hello @OoLoueLoueoO,

Can you give some code to reproduce this issue ?
How do you construct p_data_ ? This kind of exception can be raised when Data and Model mismatch.

OoLoueLoueoO, 2025-01-02:
Hi @jorisv , 

Thanks for the reply. The p_data_  = data, it is just passed down from my initialization. 

It turns out the code runs fine when I compile in the terminal with gcc but not in Simulink.
I see that Simulink is using libmwboost version of 1.83.0 while the Pinocchio is running with boost vserion 1.81.0. 
Can this be the cause of this problem?

OoLoueLoueoO, 2025-01-02:
Thanks @jorisv, the problem is resolved by fixing by defining the data after loading urdf. Hence, closing the ticket.

closed 2025-01-02 by OoLoueLoueoO

jorisv, 2025-01-03:
Good you solved your problem :)
Can you show how you fixed it ? It could help user that have the same error.

Be really careful when mixing two different version of the same library.
Boost is mostly a header only library, so you should not have so much problem.
But if you use the pinocchio serialization feature, be careful, this part use boost serialize that is a build part of boost. So building pinocchio with boost serialization 1.81 and running it with boost serialization 1.83 should give you some undefined behavior.