stack-of-tasks/pinocchio #2542: Segfault importing Pinocchio 3.3.0 Python bindings on Humble

scastro-bdai, 2025-01-03:
## Bug description

After upgrading to the ROS Humble December 20, 2024 sync, which upgrades from 2.6.21 to 3.3.0. importing the Python bindings segfaults.

## Expected behavior

Importing should not segfault.

## Reproduction steps

Steps to reproduce the behavior:
1. `python3`
2. `import pinocchio`

Before upgrade:
```
>>> import pinocchio
>>> pinocchio.__version__
'2.6.21'
```

After upgrade
```
sudo apt update && sudo apt install --upgrade ros-humble-pinocchio -y

>>> import pinocchio
Segmentation fault (core dumped)
```

### System

- OS: Ubuntu 22.04
- Pinocchio version: 3.3.0

jcarpent, 2025-01-03:
Thanks @scastro-bdai, for raising this issue. Could you show the GDB results when running the Python import?

scastro-bdai, 2025-01-03:
Sorry for the beginner question, but how do you do that in Python?

scastro-bdai, 2025-01-03:
I think I got somewhere... install Python debug symbols and ran a simple file that just runs `import pinocchio`. I get this:

```
(gdb) run test_import.py 
Starting program: /usr/bin/python test_import.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4000640 (LWP 15415)]
[New Thread 0x7ffff3600640 (LWP 15416)]
[New Thread 0x7ffff0c00640 (LWP 15417)]
[New Thread 0x7fffee200640 (LWP 15418)]
[New Thread 0x7fffeb800640 (LWP 15419)]
[New Thread 0x7fffe6e00640 (LWP 15420)]
[New Thread 0x7fffe4400640 (LWP 15421)]
[New Thread 0x7fffe1a00640 (LWP 15422)]
[New Thread 0x7fffdf000640 (LWP 15423)]
[New Thread 0x7fffde600640 (LWP 15424)]
[New Thread 0x7fffd9c00640 (LWP 15425)]

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00005555556a7df8 in _Py_Dealloc (op=<unknown at remote 0x7fffffff6dc0>) at ../Objects/object.c:2301
2301    ../Objects/object.c: No such file or directory.
```

Still seems like I'm not doing it right?

nim65s, 2025-01-03:
Thanks for the report. With `ros-humble-pinocchio-dbgsym` in a `ros:humble` docker, I get:
```cpp
#0  0x00005c23a5f77df8 in _Py_Dealloc ()
#1  0x00007add40e4f74f in _Py_DECREF () at /usr/include/python3.10/object.h:500
#2  _Py_XDECREF () at /usr/include/python3.10/object.h:567
#3  xdecref<_object> () at /usr/include/boost/python/refcount.hpp:38
#4  operator= () at /usr/include/boost/python/handle.hpp:85
#5  operator= () at /usr/include/boost/python/args_fwd.hpp:19
#6  __copy_m<boost::python::detail::keyword const*, boost::python::detail::keyword*> () at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:385
#7  __copy_move_a2<false, boost::python::detail::keyword const*, boost::python::detail::keyword*> () at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:494
#8  __copy_move_a1<false, boost::python::detail::keyword const*, boost::python::detail::keyword*> () at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:522
#9  __copy_move_a<false, boost::python::detail::keyword const*, boost::python::detail::keyword*> () at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:530
#10 copy<boost::python::detail::keyword const*, boost::python::detail::keyword*> () at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:619
#11 operator, () at /usr/include/boost/python/args.hpp:89
#12 visit<boost::python::class_<pinocchio::SE3Tpl<double, 0>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > () at ./include/pinocchio/bindings/python/spatial/se3.hpp:61
#13 0x00007add40e4dfe8 in visit<boost::python::def_visitor<pinocchio::python::SE3PythonVisitor<pinocchio::SE3Tpl<double, 0> > >, boost::python::class_<pinocchio::SE3Tpl<double, 0>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > () at /usr/include/boost/python/def_visitor.hpp:31
#14 visit<boost::python::class_<pinocchio::SE3Tpl<double, 0>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > () at /usr/include/boost/python/def_visitor.hpp:67
#15 def<pinocchio::python::SE3PythonVisitor<pinocchio::SE3Tpl<double, 0> > > () at /usr/include/boost/python/class.hpp:221
#16 expose () at ./include/pinocchio/bindings/python/spatial/se3.hpp:226
#17 0x00007add40e4de9c in exposeSE3 () at ./.obj-x86_64-linux-gnu/bindings/python/./bindings/python/spatial/expose-SE3.cpp:20
#18 0x00007add41169cbc in init_module_pinocchio_pywrap_default () at ./.obj-x86_64-linux-gnu/bindings/python/./bindings/python/module.cpp:95
#19 0x00007adde8a926e3 in boost::python::handle_exception_impl(boost::function0<void>) () from /lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
#20 0x00007adde8a93c69 in boost::python::detail::init_module(PyModuleDef&, void (*)()) () from /lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
#21 0x00005c23a60a558a in ?? ()

nim65s, 2025-01-03:
I've re-enabled https://github.com/stack-of-tasks/pinocchio/actions/workflows/ros-ci.yml

jcarpent, 2025-01-04:
@nim65s So you confirm there is an issue? If it is the case, could you share the docker image.
We will try to investigate it.

nim65s, 2025-01-04:
Yes, I'm currently working on adding the CI back, and I found another issue on eigenpy which so I won't work on this segfault right now, but a reproducer is simply:
```Dockerfile
FROM ros:humble
RUN apt update && apt install -qy ros-humble-pinocchio ros-humble-pinocchio-dbgsym gdb

jcarpent, 2025-01-04:
EigenPy might be the cause. No?

nim65s, 2025-01-04:
eigenpy seems to work fine in this docker:
```python
root@b6e2fda1be49:/# python3
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import eigenpy
>>> eigenpy.__file__
'/opt/ros/humble/lib/python3.10/site-packages/eigenpy/__init__.py'
>>> eigenpy.__version__
'3.8.2'
>>> eigenpy.Quaternion(1, 2, 3, 4).norm()
5.477225575051661

jcarpent, 2025-01-04:
I think I have found the issue. Could you please show me how to compile Pinocchio with ROS ?

danielsanjosepro, 2025-01-09:
Any updates on this? Or any quick fix for now?

jcarpent, 2025-01-09:
> Any updates on this? Or any quick fix for now?

Quick fix is certainly using ROS via robostack.

danielsanjosepro, 2025-01-09:
> > Any updates on this? Or any quick fix for now?
> 
> Quick fix is certainly using ROS via robostack.
I am working on a docker container (`ros:humble`) and my quick fix for now is to simple remove the python packages from the `site-packages` and install `pinocchio` on a virtual environment...
```bash
rm -rf /opt/ros/humble/lib/python3.10/site-packages/pinocchio
rm -rf /opt/ros/humble/lib/python3.10/site-packages/hppfcl
```
With this hack I am able to run pinocchio with python... In case anyone in this situation needs a hotfix

nim65s, 2025-01-09:
if you are using the PyPI `pin` package from pip, please not that this one is compiled with boost 1.87, so you can't mix the c++ things from there with anything compiled with boost from ros/ubuntu as it is boost 1.74

jorisv, 2025-01-10:
I think we found the issue. It's clang 14 compiler bug with Boost.Python.

I found a really close issue [here](https://discourse.llvm.org/t/apparently-broken-asm-with-boost-python-clang-14/65276).
It's patched in more recent clang version, but I don't think we can manage that on the ROS CI.

There is no issue when building it with GCC so we will try to remove this [patch](https://github.com/ros2-gbp/pinocchio-release/blob/patches/release/humble/pinocchio/0001-Patch-Use-clang.patch)

nim65s, 2025-01-10:
I will try to remove that patch, but maybe we'll get other OoM on ROS buildfarm builders.

nim65s, 2025-02-20:
We still have a lot of issues with OoM on the ROS buildfarm, even on eigenpy. But I suspect there is no isolation, so what does matter is not how much ram we use, but how much ram is used in total on the runner when it is the turn of pinocchio to build and for all that build time (so reducing the build time should also reduce the probability of OoM…)

So I tried a few things:
```
                                   time     mrss      size                   import
dist   compiler linker testing                                                     
humble clang    BFD    OFF       905.06  1913052  39793416  free(): invalid pointer
                       ON       2153.84  2660104  39793416  free(): invalid pointer
                LLD    OFF       903.69  1913040  40124672  free(): invalid pointer
                       ON       2149.97  2660632  40124672  free(): invalid pointer
                MOLD   OFF       902.49  1913224  40204608  free(): invalid pointer
                       ON       2150.73  2660492  40204608  free(): invalid pointer
       gcc      BFD    OFF      1097.41  3876244  39362056                       ok
                       ON       2711.11  6261104  39362056                       ok
                LLD    OFF      1097.67  3875980  39515416                       ok
                       ON       2716.80  6261464  39515416                       ok
jazzy  clang    BFD    OFF       926.58  1998324  40514608                       ok
                       ON       2181.69  2789412  40514608                       ok
                LLD    OFF       926.05  1997720  40745800                       ok
                       ON       2178.26  2789504  40745800                       ok
                MOLD   OFF       923.72  1997428  41348400                       ok
                       ON       2175.73  2789176  41348400                       ok
       gcc      BFD    OFF      1076.18  3540652  39552264                       ok
                       ON       2548.29  5056184  39552264                       ok
                LLD    OFF      1075.22  3540092  39607976                       ok
                       ON       2545.08  5056256  39607976                       ok
                MOLD   OFF      1074.09  3540184  41285840                       ok
                       ON       2544.12  5056380  41285840                       ok
```

So I guess we should start by just removing the test on all our packages on that buildfarm: they never caught something useful for us, even if in the case of the segfault of this issue, while they clearly should have.

I'll also avoid clang on humble, but keep it on noble (jazzy + rolling)

Also, playing with linkers does not seem relevant at all. I'm a bit surprised about this, but it is not an issue to stay on BFD :)

([here](https://gist.github.com/nim65s/b605252a05cfea5d32a07d4576e3db3b) are the scripts I used to get those results)

nim65s, 2025-02-20:
Also, I expect the situation to improve when we'll put the python bindings in another package

nim65s, 2025-03-03:
the fix has reached ros2-testing:
```
$ docker run --rm -it ros:humble
# sed -i s=/ros2/=/ros2-testing/= /etc/apt/sources.list.d/ros2-latest.list
# apt update && apt install -qy ros-humble-pinocchio
# python3 -c "import pinocchio; print(pinocchio.__version__)"
3.4.0

nim65s, 2025-04-01:
Finally fixed:
```
$ docker run --rm -it ros:humble
# apt update && apt install -qy ros-humble-pinocchio
# python3 -c "import pinocchio; print(pinocchio.__version__)"
3.4.0

closed 2025-04-01 by nim65s