stack-of-tasks/pinocchio #2552: Unsupported pickling of GeometryData

joao-pm-santos96, 2025-01-09:
## Bug description

Hello developers!
When I try to pickle an instance of GeometryData I get 

```bash
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Pickling of "pinocchio.pinocchio_pywrap_default.GeometryData" instances is not enabled (http://www.boost.org/libs/python/doc/v2/pickle.html)
```

Is this behavior expected? I would think not since the CHANGELOG says, for version 2.2.3, that "This new release provides: pickling and serialization of Data structures"

Once again, thank you!

### Code

```python
import pinocchio as pin
import pickle 

model = pin.GeometryModel()
data = model.createData()

pickle.dumps(data)
```

This piece of code returns the error described above.

### System

- OS: Ubuntu 24.04
- Pinocchio version: 3.3.1
- Python version: 3.12.8

jcarpent, 2025-01-09:
This is not yet possible. What are your needs for that?

jcarpent, 2025-01-09:
Yes, indeed, it is very easy to pickle this quantity by creating a list out of the geometry data object, as follows:
```python
active_collision_pairs_list = geom_data.activeCollisionPairs.tolist()
```
I will close this issue, as it is not something we plan to add in the near future, except if someone is willing to implement the serialization pipeline for `GeometryData`.

closed 2025-01-09 by jcarpent