stack-of-tasks/pinocchio #2559: Missing wheels for x86_64 on macOS

devbanu, 2025-01-24:
## Bug description

Previously (2.7.0 release) the macOS wheels came in two versions: arm64 and x86_64.
(Which reflects reality that there are many machines running macOS on Intel processors).

In the most recent release (3.3.1), the wheels for macOS come only in the arm64 flavour.

## Expected behavior

MacOS wheels to come with both x86_64 and arm64 versions.

## Reproduction steps

Compare files in https://pypi.org/project/pin/#files with previous release (2.7.0).

- OS: macOS
- Pinocchio version: 3.3.1

devbanu, 2025-01-24:
@nim65s I think this is relevant for you.

emrecemal, 2025-02-10:
Hi,

I think the Installation failure that I encountered with the Python 3.13 virtual environment on macOS M1 might be related to here. The main problem is `cmeel-octomap test failure`

When attempting to install `pin` via pip in a Python 3.13 virtual environment on macOS 13.6.8 (M1), the installation fails during the build of `cmeel-octomap` due to a test failure (`test_color_tree`). 

The system-installed Python 3.13 does work with Pinocchio 3.3.1, but virtual environment installation attempts fail.

### Error log
```
...
[ 68%] Linking CXX executable .../bin/normals_example
[ 93%] Built target test_color_tree
[ 94%] Building CXX object src/testing/CMakeFiles/unit_tests.dir/unit_tests.cpp.o
...
14/15 Test #14: test_color_tree ..................***Failed    0.65 sec
test failed: 3431!=1034 in .../test_color_tree.cpp, line 52
...
subprocess.CalledProcessError: Command '['cmake', '--build', ..., '-t', 'test']' returned non-zero exit status 2.
```
You can see the full terminal output: [full_terminal.txt](https://github.com/user-attachments/files/18731979/full_terminal.txt)

nim65s, 2025-02-10:
Hi @emrecemal,

Your issue is not related, because wheels for macosx_arm64 are available, while those for macosx_x86-64 are not.

Also, I don't understand your issue. Your output shows that you already have a working version of pinocchio, so what are you trying to do ? Please open another issue for that.

nim65s, 2025-02-14:
This is fixed

closed 2025-02-14 by nim65s