stack-of-tasks/pinocchio #2562: Adding Contact and Calculate Forward Dynamics

haowenlAmazon, 2025-01-28:
Dear Pinocchio devs,

I created a urdf model for a simple inverted pendulum model on two wheels (one body + two cylinders).
To get the forward dynamics, I added the contact frame for each wheel as follows:
```

    // add contact frame 
    pinocchio::FrameIndex left_wheel_id = model.getFrameId("left_wheel");
    pinocchio::FrameIndex right_wheel_id = model.getFrameId("right_wheel");

    double r = 0.1;
    pinocchio::SE3 contact_placement = pinocchio::SE3::Identity();
    contact_placement.translation() << 0, 0, -r;
    
    model.addFrame(pinocchio::Frame("left_wheel_contact_frame", 
                                    model.frames[left_wheel_id].parent,
                                    left_wheel_id,
                                    contact_placement,
                                    pinocchio::FIXED_JOINT));

    model.addFrame(pinocchio::Frame("right_wheel_contact_frame", 
                                    model.frames[right_wheel_id].parent,
                                    right_wheel_id,
                                    contact_placement,
                                    pinocchio::FIXED_JOINT));
```
The wheel contact seems to use CONTACT_3D, but I'm not sure how to construct the contact in C++ and get the forward dynamics from here. 

Thanks!

LudovicDeMatteis, 2025-01-28:
Hi @haowenlAmazon,
To construct the contact, you should use RigidConstraintModel, see [here](https://github.com/stack-of-tasks/pinocchio/blob/2e86eca560beddb178c30eb9d0174a51cd76a7ae/include/pinocchio/algorithm/contact-info.hpp#L245) for the constructor. Typically, that would require the contact type, robot model and the parent joints and relative placements of the contact points - for contacts with the floor, you can pass only on parent joint and contact placement.
Once the contact models are created, you must create the RigidConstraintData objects by simply calling `createData`.

For the dynamic, you can call `constrainedABA` or `constraintDynamics` (cf https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#aba1750befc1c37d7b92c2370a9b11371)

Please note that RigidContactModel will be extended in the future to include more contact models, including rolling contacts. (ref https://hal.science/hal-01182733/document)

haowenlAmazon, 2025-01-28:
Hi @LudovicDeMatteis,

Thank you for your response. This makes a lot of sense. 
My though is to use a frame placement directly below the wheel link to simulate the rolling wheel at each iteration. 
Closing this ticket and thanks again!

closed 2025-01-28 by haowenlAmazon