stack-of-tasks/pinocchio #2563: Test failure on crocoddyl on macos: `TypeError: No to_python (by-value) converter found for C++ type: boost::shared_ptr<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>>`

nim65s, 2025-01-29:
Hi,

crocoddyl v2.1.0 was building fine on macos with pinocchio v3.3.0, but some tests are failing with pinocchio v3.3.1:
```python
crocoddyl> The following tests FAILED:
crocoddyl>       29 - test_pybinds_actuations (Failed)
crocoddyl>       30 - test_pybinds_actions (Failed)
crocoddyl>       31 - test_pybinds_shooting (Failed)
crocoddyl>       33 - test_pybinds_costs (Failed)
crocoddyl>       34 - test_pybinds_contacts (Failed)
crocoddyl>       35 - test_pybinds_impulses (Failed)
crocoddyl>       37 - test_pybinds_copy (Failed)
```

with eg.
```python
crocoddyl> ======================================================================
crocoddyl> ERROR: test_calc (__main__.TalosArmShootingTest.test_calc)
crocoddyl> ----------------------------------------------------------------------
crocoddyl> Traceback (most recent call last):
crocoddyl>   File "…/unittest/bindings/test_shooting.py", line 30, in setUp
crocoddyl>     self.PROBLEM_DER = crocoddyl.ShootingProblem(
crocoddyl>                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
crocoddyl>   File "…/unittest/bindings/factory.py", line 445, in createData
crocoddyl>     data = DifferentialFreeFwdDynamicsDataDerived(self)
crocoddyl>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
crocoddyl>   File "…/unittest/bindings/factory.py", line 459, in __init__
crocoddyl>     self.pinocchio = pinocchio.Model.createData(model.state.pinocchio)
crocoddyl>                                                 ^^^^^^^^^^^^^^^^^^^^^
crocoddyl> TypeError: No to_python (by-value) converter found for C++ type: boost::shared_ptr<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>>
```

Reverting https://github.com/stack-of-tasks/pinocchio/pull/2469/commits/2e80bdc4581b6eb3e0f178ec82402b5b281fe8f1 fixes that issue.
Ref. #2469 #2462

cmastalli, 2025-01-29:
> I don't understand this fix. For me, when using Boost.Python, you need to declare converters support for shared_ptr, like that:
> 
> boost::python::register_ptr_to_python<boost::shared_ptr<T>>()

I agree with @jcarpent. For me, the solution should be registering the pointer, and should be done in Pinocchio to avoid doing this in each project.

jcarpent, 2025-01-29:
> For me, the solution should be registering the pointer, and should be done in Pinocchio to avoid doing this in each project.

For std::shared_ptr, yes. For boost::shared_ptr, I would say that is legacy and should be avoided.

nim65s, 2025-01-29:
I also agree :)

This issue is documenting a regression, not proposing a fix that would revert a good commit.

I think crocoddyl was missing something, but crocoddyl was working fine until v3.3.1. When pinocchio upgrade to v3.3.1 broke crocoddyl, this unveil that the pinocchio upgrade was in fact a breaking change, even if that was not expected.

We'll fix that properly in crocoddyl, and close this issue, but if anybody else has a similar problem, they will easily find this as documentation: a hotfix is to revert that commit, and the proper fix is to expose correctly the shared pointers.

closed 2025-01-30 by jcarpent