stack-of-tasks/pinocchio #2568: renaming the python module

nim65s, 2025-01-30:
Hi,

As a new major version is on its way, I would like to discuss the name of our python module.
Currently, it is almost always used as `import pinocchio as pin`, and our PyPI package is also `pin`.

According to PyPI fair-play rules, we should therefore expose a `pin` module, and only that.
Moreover, the owner of the `pinocchio` package on PyPI is not going to let that name go. So obviously, if anybody needs both pinocchio (our lib) and pinocchio (the nose testing framework addons), nothing will work as expected.

So I propose that in the next version, we move our `pinocchio` python module under the `pin` name.

A dummy `pinocchio` package with a deprecation warning and a re-export of `pin` can obviously be provided for some time to limit the number of things that will be broken by this change, as we did in the hpp-fcl -> coal renaming.

Also, I think we should keep the pinocchio name for everything else than the python module name and the PyPI package.

I'm available to handle this task if nobody is against this change.

cmastalli, 2025-01-30:
Hi, @nim65s! I still don't understand the need to do this. Could you provide a webpage where I can read more about the PyPI rules?

As a Pinocchio user, I like not to rename the module as `pin`.  I guess the reason is that `pinocchio` is such a nice name

jcarpent, 2025-01-30:
I do agree with Carlos. PyPI is one of the package manager for Python. I don't understand why it is required to stick to PyPi'rules.

nim65s, 2025-01-30:
This is not required, only a convention to help end users. 
The basic idea is "play well with others", which we are currently not doing.
https://packaging.python.org/en/latest/discussions/distribution-package-vs-import-package/ provide more details.
But I get it, you don't want to do this.

closed 2025-01-30 by nim65s