stack-of-tasks/pinocchio #2573: Add a basic C++ visualization API / upstream `Simple-Robotics/pinocchio-visualizers`

ManifoldFR, 2025-02-11:
Make sure your proposal checks out the following questions:

- [x] Is your feature request related to a problem? Please describe, *e.g.* "I'm always frustrated when..."
- [x] Describe concisely the solution you'd like.
- [x] Describe alternatives you've considered.
- [x] Do you have screenshots or sketches to illustrate your proposal?

----

A while back, there was discussion about providing basic tools for users to provide C++-based visualizers to Pinocchio, which would integrate with existing elements of the Pinocchio C++ API (computing frame and geometry placements, etc), and provide a C++ equivalent to the `pinocchio.visualizers.BaseVisualizer` Python class (although @jcarpent suggests this interface itself could/should be improved).

As a result I made the header-only library https://github.com/Simple-Robotics/pinocchio-visualizers, which also embarks a Boost.Python visitor class for exposing the C++ visualizer to Python. So far only two (very WIP) projects use it:
- [my attempt at a C++ Rerun visualizer](https://github.com/ManifoldFR/pinocchio-rerun/blob/main/src/rerun_visualizer.hpp)
- the new SDL3-based [proof-of-concept renderer](https://github.com/ManifoldFR/candlewick)

After discussion earlier today with @jcarpent we think it might be a good idea to upstream this into Pinocchio, perhaps as part of the extras CMake target.

jcarpent, 2025-02-11:
Wonderful @ManifoldFR. Thanks a lot for this nice bunch of enhancements.

ManifoldFR, 2025-02-18:
Resolved by #2574

PR #2574 (2025-02-17): Add C++ visualization API and some macros for scalar contexts
This pull request upstreams the contents of the single-header library [Simple-Robotics/pinocchio-visualizers](https://github.com/Simple-Robotics/pinocchio-visualizers), as discussed on issue #2573.
This API reproduces the behaviour of the existing Pinocchio Python bindings' `pinocchio.visualizers.BaseVisualizer` class (which is in Python).

* Add header `<pinocchio/visualizers/base-visualizer.hpp>`
* Add new CMake target `pinocchio_visualizers`
* Add namespace `pinocchio::visualizers`
* Add base class `pinocchio::visualizers::BaseVi [... cut]
closed 2025-02-18 by ManifoldFR