stack-of-tasks/pinocchio #2575: Remove example-robot-data as a submodule

jcarpent, 2025-02-12:
Dear all,

The size of the release keeps on inflating due partly to the example-robot-data submodule.
This last project will increase with time while increasing the size of the release of Pinocchio, for limiting gain.

I would suggest removing it. 

An alternative would be to only account for a few essential models to test and validate Pinocchio's.
Any comments and suggestions are welcome.

jorisv, 2025-02-13:
We can also add example-robot-data as a pinocchio build dependency.

It's not possible, because right now example-robot-data depend of pinocchio.
We could split this project into example-robot-data-resources (only the robots) and example-robot-data-scripts (python scripts and c++ headers) to allow pinocchio to depend of example-robot-data-resources.

cmastalli, 2025-02-13:
To follow up on @jorisv's suggestion, I would also keep `example-robot-data` fundamentally similar. To do so, we can add `example-robot-data-resources` and `example-robot-data-scripts` as git submodules. This won't break the API and allows Pinocchio (or other packages) to do what is more convenient.

jorisv, 2025-02-13:
I was more thinking about using example-robot-data packages and get rid of the submodule.

So example-robot-data repository will be unchanged (only some CMake options would be added to select if we want to install resources and/or scripts).

But conda-forge/nix/robotpkg/pip/etc... would provide 3 packages:
- example-robot-data: meta package that depend of example-robot-data-resources and example-robot-data-scripts
- example-robot-data-resources: no dependency
- example-robot-data-scripts: pinocchio dependency

nim65s, 2025-02-13:
I'm already doing this in robotpkg, and doing the same in all our packages has been on my todolist for quite a long time. I'll give this another try.

nim65s, 2025-07-08:
Hum, I guess I'll have to add example-robot-data to ROS too