stack-of-tasks/pinocchio #2578: Issue importing pinocchio.casadi on python venv

caro33, 2025-02-14:
Hello, due to company policy, I am unable to use conda, so I am testing the pinocchio.casadi module in a Python venv. I installed pinocchio following the procedure below, but I encountered an error. Is there a way to resolve this without using conda? The Python version I need to use is 3.8 due to the requirements of other libraries.

Install process
```
python3.8 -m venv venv
source venv/bin/activate
pip install --upgrade pip
python -m pip install -U cmeel-boost eigenpy hpp-fcl pin
python -c "import pinocchio.casadi as cpin"
```

Error message
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pinocchio.casadi'
```

nim65s, 2025-02-14:
Hi,

This require some work on the packaging of casadi (especially its dependencies) for pip first. Such packaging exist, but is not compatible with what we have in pinocchio yet. Work in this direction is planned but I can't tell when it will be ready for now.

ref. https://github.com/casadi/casadi/issues/3423

caro33, 2025-02-17:
OK I see. Thank you for your reply.

closed 2025-02-17 by caro33