stack-of-tasks/pinocchio #2587: Pinocchio does not work on a fresh python3.10

tomsch420, 2025-02-19:
## Bug description

Greetings, i am having trouble using pinocchio with python3.10.
I made a new python3.10 virtual environment, updated pip and installed pinocchio. I could not import pinocchio.
I hope this helps.

## Expected behavior

I expect to be able to import pinnocchio.

## Reproduction steps

Steps to reproduce the behavior:
```
mkvirtualenv -p python3.10 pin-test
pip install -U pip
pip install pin
python -c "import pinocchio"
```

This results in the error:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/tom_sch/.virtualenvs/pin-test/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py", line 19, in <module>
    from .pinocchio_pywrap_default import *
ImportError: /usr/lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_fopen
```

### System

- OS: Ubuntu 20.04
- Pinocchio version: 3.4.0

meijie-jesse, 2025-04-15:
If you are using conda env, then you can try 
conda install -c conda-forge pinocchio

nim65s, 2025-04-16:
This can't be reproduced, and, as already said, probably comes from a misconfigured environment.

@tomsch420: I'll close this, but feel free to ask for help in fixing this environment.

@meijie-jesse: please don't ask pip users to also use conda: this will directly break pinocchio.

closed 2025-04-16 by nim65s