stack-of-tasks/pinocchio #2592: ModuleNotFoundError installed via ros

qiyuanchn, 2025-02-23:
Hello! 
I downloaded it through sudo apt install ros-$ROS_DISTRO-pinocchio, but after running it, "import pinocchio as pin" 
ModuleNotFoundError: No module named 'pinocchio'" will appear. 
And it seems that automatic completion and highlighting cannot be done in vscode.
I am not sure what I should do next. Can you help me with this problem?

nim65s, 2025-02-24:
Hi,
Releases on ROS are in progress.
Automatic completion and highlighting won't be available for now, though.

qiyuanchn, 2025-03-03:
Thanks for your reply! I have installed the pinocchio library via robotpkg.

> Hi, Releases on ROS are in progress. Automatic completion and highlighting won't be available for now, though.

closed 2025-03-03 by qiyuanchn