stack-of-tasks/pinocchio #2600: constrainedABA not available via the python interface

hli-bdai, 2025-03-02:
Hello,

I installed pinocchio via `robotpkg` and was trying to simulate the dynamics of a closed-chain system with `constrainedABA` python. It seems that this function is not available. My pinocchio version is 3.3.1. Do I have to build from source in order to use this function in python?

jcarpent, 2025-03-03:
Thanks @hli-bdai for raising this issue.
I will handle it when I have more time.

In the meantime, you can use "constraintDynamics" that does the same computations, but with a different (more complex) approach. 
Just to inform you that with the core dev team of Pinocchio in [Willow](https://www.di.ens.fr/willow/), we are currently massively refactoring and extending the code for constrained dynamics for faster computations and extended generality (frictional contacts, joint limits, joint friction, etc.).

hli-bdai, 2025-03-03:
@jcarpent Thank you for the quick update. Are the differences between "constraintDynamics" and "constrainedABA" only in the computation complexity? Are there any differences in handling the loop constraints, for instance, constraint violation with the same number of solver iterations?