stack-of-tasks/pinocchio #2604: Inconsistency between tau and model.nv in pinocchio.forwardDynamics Function

qiyuanchn, 2025-03-03:
Hello, I'm encountering an issue while using the forwardDynamics function.
Here is my code:
```
import pinocchio as pin 
import numpy as np

urdf_path = "urdf4.urdf"
model = pin.buildModelFromUrdf(urdf_path, root_joint=pin.JointModelFreeFlyer())
data = model.createData()

print(model.nq) 
print(model.nv) 

q = pin.neutral(model)
v = np.zeros(model.nv)
tau = np.zeros(model.nv) 

a = pin.forwardDynamics(model, data, q, v, tau)
print(a)
```
When I run this code, I get the following error:
```
15
10
Traceback (most recent call last):
    a = pin.forwardDynamics(model, data, q, v, tau)
ValueError: wrong argument size: expected 10, got 15
hint: tau.size() is different from model.nv
```

Because I want to use it on a drone, I added root_joint=pin.JointModelFreeFlyer(). I've initialized tau as np.zeros(model.nv), so I'm confused why it says tau.size() is different from model.nv. The model.nv is printed as 10, and I've set tau to have the same size. Can anyone help me figure out what's going wrong here? Thanks in advance!

EtienneAr, 2025-03-03:
Hello, could you also provide your URDF to reproduce this issue ?

qiyuanchn, 2025-03-03:
Thanks for your reply. Here is my URDF `iris.urdf`.
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from iris.xacro                     | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="iris">
  <link name="iris/base_link"/>
  <joint name="iris/base_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="iris/base_link"/>
    <child link="iris/base_link_inertia"/>
  </joint>
  <link name="iris/base_link_inertia">
    <inertial>
      <mass value="1.5"/>
      <!-- [kg] -->
      <origin xyz="0 0 0"/>
      <inertia ixx="0.0347563" ixy="0.0" ixz="0.0" iyy="0.0458929" iyz="0.0" izz="0.0977"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://iris_description/meshes/iris.dae" scale="1 1 1"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.47 0.47 0.11"/>
        <!-- [m] [m] [m] -->
      </geometry>
    </collision>
  </link>
  

<link name="stand_0">
    <visual>
      <geometry>
        <cylinder length="0.2" radius="0.009"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 -0.125"/>
      <material name="blue">
        <color rgba="0 0 .8 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.2" radius="0.009"/>
      </geometry>
      
    </collision>
    <inertial>
      <mass value="0.2"/>
      <inertia ixx="0.4" ixy="0.0" ixz="0.0" iyy="0.4" iyz="0.0" izz="0.2"/>
    </inertial>
  </link>

  <joint name="stand_0_joint" type="fixed">
    <parent link="iris/base_link"/>
    <child link="stand_0"/>
    <origin xyz="0.13 -0.22 0.023"/>
  </joint> 
  
  <link name="stand_1">
    <visual>
      <geometry>
        <cylinder length="0.2" radius="0.009"/>
      </geometry>
[... 290 lines cut ...]
  <gazebo>
    <plugin filename="librotors_gazebo_motor_model.so" name="iris_back_right_motor_model">
      <robotNamespace>iris</robotNamespace>
      <jointName>iris/rotor_3_joint</jointName>
      <linkName>iris/rotor_3</linkName>
      <turningDirection>cw</turningDirection>
      <timeConstantUp>0.0125</timeConstantUp>
      <timeConstantDown>0.025</timeConstantDown>
      <maxRotVelocity>838</maxRotVelocity>
      <motorConstant>8.54858e-06</motorConstant>
      <momentConstant>0.016</momentConstant>
      <commandSubTopic>gazebo/command/motor_speed</commandSubTopic>
      <motorNumber>3</motorNumber>
      <rotorDragCoefficient>8.06428e-05</rotorDragCoefficient>
      <rollingMomentCoefficient>1e-06</rollingMomentCoefficient>
      <motorSpeedPubTopic>motor_speed/3</motorSpeedPubTopic>
      <rotorVelocitySlowdownSim>10</rotorVelocitySlowdownSim>
    </plugin>
  </gazebo>
  <gazebo reference="iris/rotor_3">
    <material>Gazebo/Red</material>
  </gazebo>
</robot>

```

EtienneAr, 2025-03-03:
With the help of @jorisv , we realized that the algorithm you might want to use is actually `aba` .

See https://github.com/stack-of-tasks/pinocchio/blob/187afafcfe22d7ac16a26241c0b13a76d04d82c1/examples/simulation-inverted-pendulum.py#L97

In the case of no contacts, the aba algorithm is doing the "forward dynamics".

Note: The forwardDynamics function that you are using is actually though in the context of contact/constrained dynamics and expect other kind of arguments, that somehow match in number the one that you provided...
See 
```python
import pinocchio
help(pinocchio.forwardsDynamics)
```
gives (and right now you are calling the second one)
```python
forwardDynamics(...)
    forwardDynamics( (Model)model, (Data)data, (numpy.ndarray)q, (numpy.ndarray)v, (numpy.ndarray)tau, (numpy.ndarray)constraint_jacobian, (numpy.ndarray)constraint_drift [, (float)damping=0]) -> numpy.ndarray :
        Solves the constrained dynamics problem with contacts, puts the result in context::Data::ddq and return it. The contact forces are stored in data.lambda_c.
        Note: internally, pinocchio.computeAllTerms is called.

    forwardDynamics( (Model)model, (Data)data, (numpy.ndarray)tau, (numpy.ndarray)constraint_jacobian, (numpy.ndarray)constraint_drift [, (float)damping=0]) -> numpy.ndarray :
        Solves the forward dynamics problem with contacts, puts the result in context::Data::ddq and return it. The contact forces are stored in data.lambda_c.
        Note: this function assumes that pinocchio.computeAllTerms has been called first.
```

Best
@EtienneAr

closed 2025-03-03 by EtienneAr

qiyuanchn, 2025-03-04:
Thank you for your timely help! My problem has been solved.

> 在[@jorisv](https://github.com/jorisv)，我们意识到您可能想要使用的算法实际上是`aba`。
> 
> 看
> 
> [pinocchio/examples/模拟倒立摆.py](https://github.com/stack-of-tasks/pinocchio/blob/187afafcfe22d7ac16a26241c0b13a76d04d82c1/examples/simulation-inverted-pendulum.py#L97)
> 
> 第 97 行 [187afaf](/stack-of-tasks/pinocchio/commit/187afafcfe22d7ac16a26241c0b13a76d04d82c1)
> 
>  a  =  pin.aba ( model , data_sim , q , v , tau_control )   #正向动力学 
> 在没有接触的情况下，aba算法正在进行“前向动力学”。
> 
> 注意：您使用的 forwardDynamics 函数实际上是在接触/约束动力学的上下文中，并期望其他类型的参数，这些参数的数量与您提供的参数数量相匹配。 请参阅
> 
> import pinocchio
> help(pinocchio.forwardsDynamics)
> 给出（现在你正在调用第二个）
> 
> forwardDynamics(...)
>     forwardDynamics( (Model)model, (Data)data, (numpy.ndarray)q, (numpy.ndarray)v, (numpy.ndarray)tau, (numpy.ndarray)constraint_jacobian, (numpy.ndarray)constraint_drift [, (float)damping=0]) -> numpy.ndarray :
>         Solves the constrained dynamics problem with contacts, puts the result in context::Data::ddq and return it. The contact forces are stored in data.lambda_c.
>         Note: internally, pinocchio.computeAllTerms is called.
> 
>     forwardDynamics( (Model)model, (Data)data, (numpy.ndarray)tau, (numpy.ndarray)constraint_jacobian, (numpy.ndarray)constraint_drift [, (float)damping=0]) -> numpy.ndarray :
>         Solves the forward dynamics problem with contacts, puts the result in context::Data::ddq and return it. The contact forces are stored in data.lambda_c.
>         Note: this function assumes that pinocchio.computeAllTerms has been called first.

qiyuanchn, 2025-03-04:
Thank you for your timely help! My problem has been solved.