stack-of-tasks/pinocchio #2605: CMake Cannot find casadi headers when compiling from source

bilal-bdai, 2025-03-03:
I am trying to build pinocchio from source to include few of the optional dependencies needed for autodiff and optimization. 

```
cmake   -DCMAKE_INSTALL_PREFIX="${INSTALL_PREFIX}"   -DCMAKE_BUILD_TYPE=Release   -DBUILD_ADVANCED_TESTING=OFF   -DBUILD_BENCHMARK=OFF   -DBUILD_DOCUMENTATION=OFF   -DBUILD_TESTING=OFF   -DBUILD_WITH_AUTODIFF_SUPPORT=ON   -DBUILD_WITH_CODEGEN_SUPPORT=ON   -DBUILD_WITH_COLLISION_SUPPORT=ON   -DBUILD_WITH_URDF_SUPPORT=ON   -DBUILD_WITH_CASADI_SUPPORT=ON   -DBUILD_PYTHON_INTERFACE=ON   -DBUILD_WITH_HPP_FCL=ON   -DPYTHON_EXECUTABLE="$(which python3)"   ..
``` 

But it seems that CMake is not able to find the casadi headers. 

```
cd /home/bhammoud/software_deps/pinocchio/build/bindings/python && /usr/bin/python3 -c "import py_compile; py_compile.compile(\"/home/bhammoud/software_deps/pinocchio/bindings/python/pinocchio/visualize/__init__.py\",\"/home/bhammoud/software_deps/pinocchio/build/bindings/python/pinocchio/visualize/__init__.pyc\")"
cd /home/bhammoud/software_deps/pinocchio/build/bindings/python && /usr/bin/cmake -E make_directory /home/bhammoud/software_deps/pinocchio/build/bindings/python/pinocchio/visualize
In file included from /home/bhammoud/software_deps/pinocchio/include/pinocchio/context/casadi.hpp:52,
                 from /home/bhammoud/software_deps/pinocchio/include/pinocchio/context.hpp:16,
                 from /home/bhammoud/software_deps/pinocchio/include/pinocchio/fwd.hpp:149,
                 from /home/bhammoud/software_deps/pinocchio/include/pinocchio/spatial/fwd.hpp:9,
                 from /home/bhammoud/software_deps/pinocchio/src/algorithm/cholesky.cpp:5:
/home/bhammoud/software_deps/pinocchio/include/pinocchio/autodiff/casadi.hpp:12:10: fatal error: casadi/casadi.hpp: No such file or directory
   12 | #include <casadi/casadi.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
cd /home/bhammoud/software_deps/pinocchio/build/bindings/python && /usr/bin/python3 -c "import py_compile; py_compile.compile(\"/home/bhammoud/software_deps/pinocchio/bindings/python/pinocchio/visualize/base_visualizer.py\",\"/home/bhammoud/software_deps/pinocchio/build/bindings/python/pinocchio/visualize/base_visualizer.pyc\")"
make[2]: *** [src/CMakeFiles/pinocchio_casadi.dir/build.make:79: src/CMakeFiles/pinocchio_casadi.dir/algorithm/cholesky.cpp.o] Error 1
make[2]: Leaving directory '/home/bhammoud/software_deps/pinocchio/build'
make[1]: *** [CMakeFiles/Makefile2:1889: src/CMakeFiles/pinocchio_casadi.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] Building CXX object src/CMakeFiles/pinocchio_cppad.dir/algorithm/impulse-dynamics.cpp.o
```

When i check if pkg config can find casadi and its headers, this is what i get 

```
WS14:~/software_deps/pinocchio/build$ pkg-config --modversion casadi
3.6.7+
WS14:~/software_deps/pinocchio/build$ pkg-config --cflags casadi
-I/opt/modeling_libraries/include
WS14:~/software_deps/pinocchio/build$ pkg-config --libs casadi
-L/opt/modeling_libraries/lib -lcasadi
WS14:~/software_deps/pinocchio/build$ ls /opt/modeling_libraries/include/casadi/
casadi_c.h  casadi.hpp  casadi.i  casadi_numpy.hpp  config.h  core  doc.i  doc_merged.i  mem.h  valgrind-casadi.supp  valgrind-python.supp
```

To investigate further i created a simple c++ file 
```
#include <casadi/casadi.hpp>
#include <iostream>

int main(){
    std::cout << "include works" << std::endl; 
    return 0; 
}

``` 

and compiled using `g++ test.cpp $(pkg-config --cflags --libs casadi) -o test`  and that seems to work just fine. 

can you recommend further steps to debug?

jcarpent, 2025-03-03:
@bilal-bdai Thanks for raising this issue.
Could you share a docker image reproducing your bug?

In the meantime, I invite you to rely on conda directly that provides the full support of Pinocchio+CasADi on Windows, Mac, and Linux.

jcarpent, 2025-03-03:
What is your version of Pinocchio?