stack-of-tasks/pinocchio #2609: Missing Quaternion Model in Pinocchio.casadi

qiyuanchn, 2025-03-07:
Hello! I've been using `pin.Quaternion` to handle quaternions. Now, I'm trying to use `pinocchio.casadi` to perform the same function, but I've found that there is `no Quaternion model in cpin`. Could you please offer some assistance?

jcarpent, 2025-03-07:
Thanks for raising this question. Indeed, we need to rewrite a part of Eigen::Quaternion class to support operations with CasADi types. I would suggest using directly SE3 objects to store your rotation quantities and play with the associated algebra.

I hope it makes sense.
Best,
Justin

qiyuanchn, 2025-03-09:
Thanks for your prompt reply! I'll try the method you mentioned.

qiyuanchn, 2025-04-23:
Hello! Following your previous suggestion, I used `M = pinocchio.casadi.XYZQUATToSE3(q)` to construct `se(3)` to solve the quaternion problem. My research object is a free-flyer. I want to transform its velocity in the `world frame` to the `body frame` using a rotation matrix. Can I directly perform the transformation using `v_body = M.rotation.T @ v_world`?