stack-of-tasks/pinocchio #2619: Pinocchio 3.2.0 not working in python 3.10, python process being killed

XiangningLin, 2025-03-16:
## Bug description

I'm using Mac OS 15.0, and I need to use Pinocchio 3.2.0 and python 3.10 in my project. I'm using conda virtual env.

I used 

> conda install pinocchio=3.2.0 -c conda-forge

and then used my python3 and try to import pinocchio and print its version. But my python process directly gets killed.

## Expected behavior

It should import pinocchio successfully and print the correct version 3.2.0

## Reproduction steps

Steps to reproduce the behavior:
1. create a conda env
> conda create --name pinocchio_test

2. get inside the env
> conda activate pinocchio_test

3. run conda to install pinocchio
>  conda install -c conda-forge pinocchio=3.2.0 -y

4. run python
> python3 -c "import pinocchio; print(pinocchio.\_\_version\_\_)"

6. the python process is being killed

### Screenshots
[image]

My system is not out of memory, and if I run other .py file without using pinocchio, it works just like the "hello world".

### System

- OS: MacOS 15.0
- Pinocchio version: 3.2.0

jcarpent, 2025-03-16:
I can reproduce the issue. Why do you need Pinocchio 3.2.0?
Could you test with Pinocchio 3.4.0 directly?

XiangningLin, 2025-03-16:
I tested with 3.4.0 directly, it has the same issue.

jcarpent, 2025-03-17:
The issue seems to come from a fresh conda environment. I've triggered a new build on conda: https://github.com/conda-forge/pinocchio-feedstock/pull/140 🤞

jorisv, 2025-03-17:
Hello @XiangningLin,

Can you give us the result of `conda list` and `conda info` in your pinocchio_test environment ?

jcarpent, 2025-03-17:
One quick solution is to create a new environment from the attached environment file using:
`conda create --file=env.txt `

[env.txt](https://github.com/user-attachments/files/19286211/env.txt)

jcarpent, 2025-03-18:
@traversaro Are you aware of any recent issues on Conda on ARM64 architecture? Currently, when installing Pinocchio in a fresh environment, we have a segfault on mac OS X ARM64 systems.

cmastalli, 2025-03-18:
@jcarpent and @traversaro, one of my students noticed this recently, including Crocoddyl, as well. However, I did not entirely think it was an error from Conda.

jorisv, 2025-03-18:
Here the `conda list` result for a pinocchio fresh install [crash.txt](https://github.com/user-attachments/files/19319020/crash.txt) and the env.txt send by @jcarpent [work.txt](https://github.com/user-attachments/files/19319021/work.txt)

I tried to create a new environment with `pinocchio==3.3.1=py312hf720c60_0` and `python 3.12.8` but it's still crashing.

olivier-roussel, 2025-03-18:
Looks like the update of `urdfdom` from 4.0.1-h922ef61_0 to 4.0.1-h1874252_1 causes the problem

olivier-roussel, 2025-03-18:
@traversaro Any idea of what could have caused that in https://github.com/conda-forge/urdfdom-feedstock/pull/34 ?

XiangningLin, 2025-03-18:
> caused that in conda-forge/urdfdom-feedstock#34

I’m able to solve the problem by using the previous urdfdom build.Thank you
guys!

traversaro, 2025-03-18:
Sorry, I was able to come back to this only this evening, and I do not have a `osx-arm64` machine to test on until next morning. I tried to reproduce the issue in GitHub Actions in https://github.com/traversaro/pinocchio-urdfdom-debugging, but `python3 -c  "import pinocchio; print(pinocchio.__version__)"` always works even on a huge matrix of different versions of pinocchio, urdfdom and python (see https://github.com/traversaro/pinocchio-urdfdom-debugging/actions/runs/13932712133). I wonder what is the difference between all your machines and GitHub Actions? Perhaps the use of zsh? Are you installing pinocchio with conda, micromamba or pixi?

XiangningLin, 2025-03-18:
I’m installing using conda environment and I’m using zsh as well. Thanks!

feixyz10, 2025-03-19:
`conda install pinocchio -c conda-forge`
`conda install urdfdom==4.0.0 -c conda-forge`
can solve my same problem.

jorisv, 2025-03-19:
I have tried with Pixi and I have the same issue.
I tested with zsh. It also crash with bash, but I launched it from zsh so maybe the test is not well done…

olivier-roussel, 2025-03-19:
We had the bug locally using conda (and @jorisv just said also with pixi). We were also using zsh.
I tried to continue your debugging (thanks for this) but could manage to make work zsh and conda properly on the CI only with micromamba:
https://github.com/olivier-roussel/pinocchio-urdfdom-debugging/actions/runs/13943549258/job/39025436549
So, with micromamba, we do not reproduce the bug on the CI, neither on bash or zsh.
Would be nice to be able to try with conda + zsh on the CI, but had troubles with setup-miniconda and conda initialization for zsh.

olivier-roussel, 2025-03-19:
We also thought that OS version differs between CI macOS runner and our local macOS arm64 machine, maybe this could be related ?
CI runner version (runs fine):
`macOS 14.7.4 23H420`
Our local macOS machine (crashes):
`macOS 15.2 24C101`

traversaro, 2025-03-19:
I am able to reproduce the problem on a physical machine with macOS 14.6.1 . I also tried tor run under lldb, and the problem does not occur, as previously reported.

traversaro, 2025-03-19:
dmesg | grep urdfdom seems to report some relevant errors.

```
[9803688.180209]: Waking up reference: 1932Thread waiting on reference 1932 woke upevaluation result: 1932, library, allowed, cache,  1742385534, 0, 35dd7095f57345b2, /Users/icub/miniforge3/envs/pinarm/lib/libboost_filesystem.dylibAMFI: '/Users/icub/miniforge3/envs/pinarm/lib/liburdfdom_model.4.0.dylib' has no CMS blob?
[9803688.181162]: AMFI: '/Users/icub/miniforge3/envs/pinarm/lib/liburdfdom_model.4.0.dylib': Unrecoverable CT signature issue, bailing out.
[9803688.314024]: Waking up reference: 1933Thread waiting on reference 1933 woke upevaluation result: 1933, library, allowed, cache,  1742385534, 8, 35dd7095f57345b2, /Users/icub/miniforge3/envs/pinarm/lib/liburdfdom_model.4.0.dylibCODE SIGNING: cs_invalid_page(0x102bfc000): p=79436[python3.13] final status 0x23000200, denying page sending SIGKILL
[9803688.314209]: CODE SIGNING: process 79436[python3.13]: rejecting invalid page at address 0x102bfc000 from offset 0x0 in file "/Users/icub/miniforge3/envs/pinarm/lib/liburdfdom_model.4.0.dylib" (cs_mtime:1741340741.0 == mtime:1741340741.0) (signed:1 validated:1 tainted:1 nx:0 wpmapped:0 dirty:0 depth:0)
```

traversaro, 2025-03-19:
I suspect a rattler-build-related regression. I think a reasonable low-cost attempt is to just rebuild urdfdom, let me try.

traversaro, 2025-03-19:
Related issues:
* https://github.com/conda-forge/python-feedstock/issues/544
* https://github.com/conda-forge/openjdk-feedstock/issues/109
* https://github.com/conda-forge/libssh2-feedstock/issues/46
* https://github.com/conda-forge/miniforge/issues/197

traversaro, 2025-03-19:
I just tried with `conda install "urdfdom==4.0.1=*_2"` and the issue seems solved, @jcarpent @jorisv @olivier-roussel @feixyz10 @XiangningLin @cmastalli can you validate this? 

Thanks for reporting the issue, this was a tricky problem that was difficult to catch as it did not occur in CI.

olivier-roussel, 2025-03-19:
I justed tested as well and it fixes the problem on our side. Thanks a lot @traversaro ! It was indeed a tricky one.

traversaro, 2025-03-19:
fyi @wolfv we were having a:

~~~
CODE SIGNING: process 79436[python3.13]: rejecting invalid page at address 0x102bfc000 from offset 0x0 in file "/Users/icub/miniforge3/envs/pinarm/lib/liburdfdom_model.4.0.dylib" (cs_mtime:1741340741.0 == mtime:1741340741.0) (signed:1 validated:1 tainted:1 nx:0 wpmapped:0 dirty:0 depth:0)
~~~

error on a build of urdfdom done with rattler-build 0.38.0, that went away when re-rendering and using rattler-build 0.39.0 . The error was quite tricky as it was not happening on `osx-64` (only on `osx-arm64` ) and was not happening on CI. Do you think this problem could still be there for other libraries, and it could make sense to look for other `osx-arm64` built with rattler-build 0.38.0 to understand if they are affected by this?

jorisv, 2025-03-19:
Thanks @traversaro !

traversaro, 2025-03-19:
And by the way, thanks @wolfv for the fast fix in https://github.com/prefix-dev/rattler-build/pull/1479 !

traversaro, 2025-03-19:
> Do you think this problem could still be there for other libraries, and it could make sense to look for other `osx-arm64` built with rattler-build 0.38.0 to understand if they are affected by this?

I could not easily find the packages built with rattler-build 0.38.0, but instead I looked at the osx-arm64 pkgs that were built from the 5/3/2025 to the 12/3/2025 (the period in which rattler-build 0.38.0 was used), see https://gist.github.com/traversaro/78ece56d217f6fca41c753a6b896de3d . Scrolling that list and looking on https://conda-metadata-app.streamlit.app/?q=conda-forge%2Flinux-64%2Fgz-math8-python-8.1.1-py39h32ec107_1.conda which one were built by rattler-build 0.38, I already found another package that we use affected by this, gz-math8-python, that also fails. Anyhow, this is becoming definitely OT here, so I will open an issue in rattler-build feedstock.

traversaro, 2025-03-19:
> I will open an issue

Opened in https://github.com/conda-forge/rattler-build-feedstock/issues/149 .

traversaro, 2025-03-20:
As the problem was solved for pinocchio and now we are tracking the problem at the conda-forge level at https://github.com/conda-forge/rattler-build-feedstock/issues/149, I guess we can close this issue?

closed 2025-03-20 by jorisv

jcarpent, 2025-03-20:
@traversaro Thanks again for the very deep investigation. You deserve a nice dinner (with us) in Paris ;)

traversaro, 2025-03-20:
> [@traversaro](https://github.com/traversaro) Thanks again for the very deep investigation. You deserve a nice dinner (with us) in Paris ;)

I should definitely pass by Paris at some point. : )