stack-of-tasks/pinocchio #2625: in 039ec083: macro PINOCCHIO_DEFAULT_QUATERNION_NORM_TOLERANCE_VALUE diappeared

florent-lamiraux, 2025-03-20:
## Bug description

I get a compilation error when using macro PINOCCHIO_DEFAULT_QUATERNION_NORM_TOLERANCE_VALUE.

## Expected behavior

No compilation error

## Additional context

I open this issue to discuss the relevance of reintroducing the macro to avoid a breaking change.
Although it seems that replacing it with somethink like `Eigen::NumTraits<typename VectorLike::Scalar>::dummy_precision()` should fix the issue.

### System

- Ubuntu-24/04
- Pinocchio version: devel branch

jorisv, 2025-03-20:
Done in this PR: https://github.com/stack-of-tasks/pinocchio/pull/2608

@cmastalli, @jcarpent any reason to remove it ?
Did we consider it public API ?

jorisv, 2025-03-20:
After some internal talk we will reintroduce PINOCCHIO_DEFAULT_QUATERNION_NORM_TOLERANCE_VALUE to the public API.

jcarpent, 2025-03-20:
Solved via #2626

PR #2626 (2025-03-20): Set back PINOCCHIO_DEFAULT_QUATERNION_NORM_TOLERANCE_VALUE definition
Close #2625
closed 2025-03-20 by jcarpent