stack-of-tasks/pinocchio #2628: .md files in doc/* not rendering math

kushal-a, 2025-03-21:
Hi,
I am new to open source dev and pinocchio and have just begun using it couple of weeks ago, so parden my naiveness. I could not help but notice that the .md files in all the folders in the doc folder are not rendering properly on github or locally. I have checked that there are issues with references as well as the math. Is this a problem or is it that way for some reason? Can someone explain or let me know if it is an issue?
Thanks

jcarpent, 2025-03-21:
The doc is accessible here: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/

kushal-a, 2025-03-21:
Yes, I understand. I was curious why the .md in the repo did not read the same way, thank you very much. Should I send a PR to fix the files on the repo?

Additionally couple of sections like Rigid Bodies/Kinematics and Rogod Bodies/Dynamics are empty in the documentation. Can I find a reference to those somewhere?

jcarpent, 2025-03-21:
> Yes, I understand. I was curious why the .md in the repo did not read the same way, thank you very much. Should I send a PR to fix the files on the repo?

I don’t see directly what you mean, but feel free to make some suggestions of improvement. Your contribution to Pinocchio is very welcome.

> Additionally couple of sections like Rigid Bodies/Kinematics and Rogod Bodies/Dynamics are empty in the documentation. Can I find a reference to those somewhere?

I suggest reading the nice book on Rigid body dynamics by Roy Featherstone.

kushal-a, 2025-03-21:
> > Yes, I understand. I was curious why the .md in the repo did not read the same way, thank you very much. Should I send a PR to fix the files on the repo?
> 
> I don’t see directly what you mean, but feel free to make some suggestions of improvement. Your contribution to Pinocchio is very welcome.

What I mean is apart from getpettoweb documentation, when one opens the .md file from those folder, either on github or locally, they be able to see the rendered markdown with proper math equations and references. 

I guess the getpettoweb documentation was made from an older version of the repo?

jcarpent, 2025-03-21:
The C++ doxygen directly parses the md files to generate the html pages.

kushal-a, 2025-03-21:
Okay, got it. Doxygen requires equations to be enclosed in \f$ tags whereas markdown requires them to be in $ tags for rendering. Markdown cannot render \f$ tags and hence the equations appear the way they do in markdown. The only way to fix it would be to make a copy of the files altogether for doxygen and markdown but I guess that is a bit redundant.

Can you suggest resources (apart from getpettoweb and Roy Featherstone) to understand the core functionality of pinocchio and documentation for its python functions, if available?

Thanks, I will close the issue as no modification is needed.

closed 2025-03-21 by kushal-a

jcarpent, 2025-03-21:
I would suggest reading our introduction paper on Pinocchio: https://hal.science/hal-01866228.
Inside, you will find further references, less demanding than reading the whole book of Roy Featherstone ;)